bblanchon/ArduinoJson #708: Throws errors when Particle Photon firmware is version 0.6.3

JerseyTechGuy, 2018-03-17:
Particle Photon Firmware version 0.6.3

Errors are as follows:

lib/ArduinoJson/ArduinoJson/Deserialization/JsonParser.hpp:74:61: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<String, void>'
lib/ArduinoJson/ArduinoJson/Deserialization/JsonParser.hpp:75:50: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<String, void>'
lib/ArduinoJson/ArduinoJson/JsonBufferBase.hpp:85:60: no matching function for call to 'makeParser(ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>*, String&, uint8_t&)'
lib/ArduinoJson/ArduinoJson/JsonObject.hpp:296:67: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const String&, void>'
lib/ArduinoJson/ArduinoJson/Data/ValueSetter.hpp:19:17: no match for 'operator=' (operand types are 'ArduinoJson::JsonVariant' and 'const String')
lib/ArduinoJson/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:144:13: 'class String' has no member named 'print'
lib/ArduinoJson/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:141:13: 'class String' has no member named 'print'

bblanchon, 2018-03-17:
Hi @JerseyTechGuy,

Can you try to set `ARDUINOJSON_ENABLE_ARDUINO_STRING` to 1, like this:

```c++
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1
#include <ArduinoJson.h>
```

Regards,
Benoit

bblanchon, 2018-03-17:
PS: From what I understand, there is an "Arduino Compatibility" header for Particle.
Maybe you just need to include it before ArduinoJson:

```c++
#include <Arduino.h>
#include <ArduinoJson.h>
```

JerseyTechGuy, 2018-03-17:
Both methods got rid of the errors however created errors with SdFat which I'm also referencing.  Good to know this solved at least the ArduinoJson issue.  Thanks!

bblanchon, 2018-03-18:
You're welcome 😄 
Thank you for using ArduinoJson.

closed 2018-03-18 by bblanchon