bblanchon/ArduinoJson #710: DynamicJsonBuffer(JSON_OBJECT_SIZE()) is limited to 10 Objects ?

yacinees, 2018-03-22:
I have read the FAQ and coulnd't find a solution to my issue.
I try to serialize data with ArduinoJson. But when I add more than 10 of it, the Serial only print the 9 and print the 10th but without his value.
Is there a limitation in the memory of a DynamicJsonBuffer.

Below there is my code

This is at the initialization of my class
`DynamicJsonBuffer *_jsonBuffer;
_jsonBuffer = new DynamicJsonBuffer(JSON_OBJECT_SIZE(11));
`

And there when I want to serialize (I browse a map to fullfil my JsonObject and after I add a timestamp
`	_jsonBuffer->clear();
	JsonVariant jsonObject = _jsonBuffer->createObject(); //reference to the JsonObject created

	//Browsing the hole map
	std::map<const char*, float>::iterator it = objectMap.begin();
	while (it != objectMap.end()) {
		jsonObject[it->first] = it->second;
		++it;
	}

	//If RTC is detected, add a timestamp to the JSONObject
	if (rtc.getPresence()) {
		jsonObject["TIME"] = rtc.getDate();
	}`

These code works fine when I have a map of 8 elements (or less) but when there is more it print this without the timestamp:
{
  "TEMPERATURE": 25.83,
  "HUMIDITY": 28.2,
  "PRESSURE": 979.49,
  "CO2": 409.21,
  "CO": 4620,
  "NH4": 298.4,
  "ETHANOL": 300.2,
  "ACETONE": 147.75,
  "BENZENE": 186.91,
  "TIME": 
}

bblanchon, 2018-03-23:
Hi @yacinees,

I believe that `rtc.getDate()` returns a `String` or an `std::string`, which forces ArduinoJson to make a copy.
Therefore, you need to increase the capacity of the `JsonBuffer` to make room for this string.
From the example you gave, 20 bytes are sufficient.

See:

* [JsonObject::set()](https://arduinojson.org/api/jsonobject/set/?utm_source=github&utm_medium=issues)
* [FAQ: How to determine the buffer size?](https://arduinojson.org/faq/how-to-determine-the-buffer-size/?utm_source=github&utm_medium=issues)
* [ArduinoJson Assistant](https://arduinojson.org/assistant/?utm_source=github&utm_medium=issues)

Regards,
Benoit

yacinees, 2018-03-23:
Hi Benoit,
Thanks for your quick answer.
Indeed, at first it returned a String but to decrease the memory taken `rtc.getDate()` returns char* now.
This is my method:
```
char* RTC::getDate() {

	DateTime now = _rtc->now();

	time_t rawtime;
	struct tm *timeinfo;

	time(0);
	timeinfo = localtime(&rawtime);
	timeinfo->tm_year = now.year() - 1900; //years are counted since 1900 in struc tm
	timeinfo->tm_mon = now.month() - 1; //months are counted between 0 and 11 in struc tm
	timeinfo->tm_mday = now.day();
	timeinfo->tm_wday = now.dayOfTheWeek();
	timeinfo->tm_hour = now.hour();
	timeinfo->tm_min = now.minute();
	timeinfo->tm_sec = now.second();

	//return timeinfo;
	return isotime(timeinfo);
}
```

**When I put 20 bytes for the buffer it works !! Many thanks for your help !!**

Though I've got a question, how did you know that 20 bytes were necessary in thiscase ? When I use ArduinoJsonAssistant it only tells me to put `JSON_OBJECT_SIZE()` at 10 and when I put JSON_OBJECT_SIZE() at 15 for example there was still the problem (I think it was because it is calculated according to the number of Objects put in the Json but actually I don't know)
I'm a little confused about how to determine the buffer size I've got to put.
In the future I will add more parameters and if I've got this same problem is there a way to determine the buffer size that I've got to put ?
Is there a limitation in the capacity of a `DynamicJsonBuffer ` ?

Again I'm really grateful for your help and your library is really really good !

Regards,
Yacine

yacinees, 2018-03-23:
I think I have understand better how to manage the Buffer. As my English is medium I didn't get all the articles and the JsonAssistant helped me to compute more precisely the capacity of the Buffer for my JSON.

I 'm still wondering if there is a limitation for the JsonBuffer or if the only limitation is the RAM of the Arduino ?

Thanks for your help.

Yacine

bblanchon, 2018-03-24:
Hi Yacine,

I knew 20 bytes were needed to store the string `"2018-03-21 22:06:01"` because it's 19 characters long plus one byte for the terminator.
Since ArduinoJson 5.13, `char*` are duplicated too; if you don't want duplication, you need to use a `const char*`.
The ArduinoJson Assistance uses a string literal, which has the type `const char*`, so there is no duplication.

ArduinoJson doesn't impose any limitation on the size of the `JsonBuffer`.
However, a `StaticJsonBuffer` is limited by the available stack memory, and a `DynamicJsonBuffer` is limited by the available heap memory.

If none of this makes sense, I invite you to read the "Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues); it explains what stack and heap are, and what are the different kinds of strings in C++.

Regards,
Benoit

yacinees, 2018-03-24:
Hi Benoit,

Thank you for your precise and instructive answer.
I think I've got a better understand about the Static and Dynamic Buffer with the stack and heap memory.
My though is that I have reached the whole capacity of it because I had a little code to test the RAM available when I do those operations and there is none.

I'm again really grateful about your library and want to congrat you for your really good job !

Regards,
Yacine

bblanchon, 2018-03-25:
You're welcome Yacine.
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2018-03-25 by bblanchon