bblanchon/ArduinoJson #711: Verifying JSON parsability without creating buffer

marecl, 2018-03-22:
TLDR: code below is used to save config JSON received from another device.  
I'm wondering if there is a way to verify JSON *without* creating JsonBuffer.  
Also is this way of handling JSON correctly?  
  
Arduino 1.8.5   
Device: ESP-01  
Board version 2.4.0-rc1  
ArduinoJson 5.13.1  
  
`````
void setup(){
  String *xxx = new String();
  *xxx = "[{\"XDDDD\":true},{\"Wololo\":false}]"; //JSON got from remote device
  bool wut = handleRequest(xxx);
  delete xxx;
}

void loop(){}

bool handleRequest(String *_in) {
  bool succ = false;
  uint8_t buffSize = _in->length() + 10;
  StaticJsonBuffer buf(buffSize);
  JsonArray & lel = buf.parseArray(*_in);
  if (lel.success()){ //here is saving
    Serial.println("Good JSON");
    succ = true;
  } else Serial.println("Bad JSON");

  return succ;
}
`````

bblanchon, 2018-03-23:
Hi @marecl,

No, there is no way to validate a JSON document without parsing it.

Remarks on the program above:

* The computation of the buffer size is incorrect, see [How to determine the buffer size?](https://arduinojson.org/faq/how-to-determine-the-buffer-size/https://arduinojson.org/faq/how-to-determine-the-buffer-size/?utm_source=github&utm_medium=issues)
* Allocating the `String` with `new` doesn't make sense, simply declare the string as a local variable.
* Passing the `String` by pointer is unusual, prefer a `const&`

Regards,
Benoit

PS: You could benefit from the "Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues); it explains what pointer and reference are, what are the different string types and how memory is managed in C++.

marecl, 2018-03-23:
@bblanchon Sorry, my question wasn't precise. Do I have to create JsonBuffer for parsing or is there a method which will just scan string and tell me if it's bad or good.  

Thanks for recommendations!  
Talking about books, I already have one but I still must practice using pointers and stuff.

bblanchon, 2018-03-24:
@marecl You do need to create a `JsonBuffer`.

closed 2018-03-24 by marecl