bblanchon/ArduinoJson #717: Print (and publish via MQTT) JSONArray

xsirpreisx, 2018-04-01:
Hi,

I am trying to create a JSONArray to store bluetooth MAC addresses and their signals strengths (RSSI) on my ESP32. I do get an array, but somehow 

```
  DynamicJsonBuffer jsonBuffer;
  JsonArray& array = jsonBuffer.createArray();

  BLEScanResults scanResults = pBLEScan->start(30);

  for (int i=0; i < scanResults.getCount(); i++) {
    BLEAdvertisedDevice device = scanResults.getDevice(i);

    BLEAddress *deviceAddress;
    deviceAddress = new BLEAddress(device.getAddress());

    JsonObject& btDevice = array.createNestedObject();
    btDevice[deviceAddress->toString().c_str()] = String(device.getRSSI()).c_str();

    Serial.print("TEST: ");
    btDevice.printTo(Serial);
    Serial.println("");
  }
  
  Serial.println("TEST Print of array: ");
  array.printTo(Serial);
```

I do get the following in the serial console/monitor:

```
TEST: {"f4:f5:e8:56:fe:db":"-53"}
TEST: {"4d:c7:82:8d:79:d1":"-50"}
TEST: {"62:27:05:87:0e:eb":"-67"}
TEST: {"7c:2f:80:97:0f:ab":"-67"}
TEST: {"56:22:38:c0:38:6f":"-89"}
TEST: {"7c:2f:80:97:0f:f5":"-85"}
TEST Print of array: 
[{"b'⸮⸮xV⸮⸮d:⸮⸮⸮?⸮⸮⸮?":"Mǂ⸮y⸮xV⸮⸮"},{"b'⸮⸮xV⸮⸮d:⸮⸮⸮?⸮⸮⸮?":"4⸮⸮⸮xV⸮⸮"},{"":"|/⸮⸮⸮xV⸮⸮"},{"|/⸮⸮⸮xV⸮⸮":""},{"7c:2f:80:97:0f:f5":"-85"},{"7c:2f:80:97:0f:f5":"-85"}]
```

So, printing the JsonObjects each does work, but when putting them in the array or trying to print that array, I do get a lot of cryptic stuff. What am I doing wrong?

xsirpreisx, 2018-04-01:
Nevermind, guess I have not used pointers, references and stuff like that for 15 years :-D

closed 2018-04-01 by xsirpreisx

bblanchon, 2018-04-03:
Hi @xsirpreisx,

Just get rid of `.c_str()` so that ArduinoJson knows it has to make a copy.

See:

* [FAQ: Why does the generated JSON document contain garbage?](http://arduinojson.org/faq/why-does-the-generated-json-contain-garbage/?utm_source=github&utm_medium=issues)

Regards,
Benoit