bblanchon/ArduinoJson #720: Not parsing, tried everything in FAQ

lolimpol, 2018-04-15:
I have tried all the solutions here:

https://arduinojson.org/faq/why-parsing-fails/?utm_source=github&utm_medium=issues

but it still doesn't work for some reason (quite a noob here).
I am using an adapted version of this code:

https://github.com/DotNetDann/ESP8266-MQTT-JSON-SK6812RGBW-HomeAssistant

to work with an NRF24L01. and an Arduino Uno, but eventuallt with an ESP8266 (Wemos D1 mini pro)
since this part works, please no comments on my terrible implementation of it.

now I simply have a char[119] called jsone, instead of json, and I even tried using the gps code in the arduinojson sample, but it crashed.

so since I'm a noob and really new to the github scene, here is the documents: Neopixel effects, authentication (not used) and the code in question. PS: the only part that matters is bool processJson.

[ESP_RF24_Digital_LEDs.zip](https://github.com/bblanchon/ArduinoJson/files/1912626/ESP_RF24_Digital_LEDs.zip)

MESA0511, 2018-04-18:
Hello bblanchon, sorry my english not very well. I use Arduino Yún parsing jsonarray and Arduino IDE version (1.8.5) this is my project: if I upload this code, I get "parsing failed!". I want to know why? Thank you!

#include <Bridge.h>
#include <HttpClient.h>
#include <Console.h>
#include <ArduinoJson.h>

void setup() {

  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);
  
  Bridge.begin();
  
  digitalWrite(13, HIGH);

  Console.begin();
  while (!Console);
}

void loop() {
  HttpClient client;
  
  client.get("http://opendata.epa.gov.tw/ws/Data/ATM00625/?$format=json");

  while (client.available()) {

    const size_t bufferSize = JSON_ARRAY_SIZE(76) + 76*JSON_OBJECT_SIZE(5) + 6420;
    DynamicJsonBuffer jsonBuffer(bufferSize);

    JsonArray& root = jsonBuffer.parseArray(client);

    if (!root.success()) {
      Console.println("Parsing Failed!");
      return;
      } else {
        Console.println("OK");
        return;
      }

  }
  
  Console.flush();

  delay(5000);
}

lolimpol, 2018-04-18:
@MESA0511, please post your own issue! Then you can have full attention of all the troubleshooters.

bblanchon, 2018-04-19:
@lolimpol Did you manage to make some progress?

lolimpol, 2018-04-19:
@bblanchon sadly not yet, I have no clue what could be wrong, I have isolated all the problematic factors out... At least I think.

bblanchon, 2018-04-20:
Can you send a **short** program that reproduces the issue (50 lines max)?

lolimpol, 2018-04-20:
@bblanchon yes, here it is, but this one runs perfectly!

[Arduino code.zip](https://github.com/bblanchon/ArduinoJson/files/1933604/Arduino.code.zip)

so there must be something in the lines I deleted between the two that is causing it not to parse, it's not the RF24 library though, as it is included here.

bblanchon, 2018-04-21:
Hi @lolimpol,

You are using an Arduino Uno, right?
The Uno has only 2KB of RAM, so it's possible that you're running out of memory.
If you want to keep the Uno, you need to reduce memory consumption.

First, you must reduce the global variables.
For example, the three following variables are wasting 663 bytes, that's 32% of the total RAM!

```c++
char text[32] = "";
char json[512] = "";
char jsone[119] = "{\"id\":1, \"state\"
```

Then, I strongly advise against using dynamic memory allocation on the Uno because it causes fragmentation.
In particular, you should not use any `String`.
Unfortunately, you cannot be sure that the libraries that you include are following this discipline.

Regards,
Benoit

PS: If none of that makes sense, please read "The missing C++ course" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

lolimpol, 2018-04-21:
@bblanchon, that's probably it. But if it can't even run on a UNO, then it won't run on a Nano, so I guess I will have to search for a more lightweight solution. But thanks for the help!

closed 2018-04-21 by lolimpol