bblanchon/ArduinoJson #725: Exception handling on esp8266 ?

DasVoelkel, 2018-04-26:
With regards to the support for the esp8266 module.
Works great in the Arduino IDE, but the esp8266 library works in eclipse too as they have a plugin for it. Sadly in the other editor this library throws 
```
error: exception handling disabled, use -fexceptions to enable
```
My Question would be if this is because of your library being incompatible with eclipse / the esp8266 plugin (which i doubt, but i gotta ask because it's stopping me from working. 
Or if this is a setting in eclipse.

If it is the latter, i am new to eclipse, does anyone know how to set this?

DasVoelkel, 2018-04-28:
I've added the paths that were missing to:
C/C++ General -> Paths and Symbols -> Source Location -> "Source folders on build path" -> "/ArduionoJson" 
And that are filter settings where the list looks and seems the same. sadly the error persists.

bblanchon, 2018-04-30:
As an alternative, you can download the single file distribution.
It's just a single `.h` that you copy in your project folder.

See:

* [Installation](https://arduinojson.org/doc/installation/?utm_source=github&utm_medium=issues)