bblanchon/ArduinoJson #731: PlatformIO: using ArduinoJson with native platform

adolfogc, 2018-05-06:
Hello, I have an Arduino project for which I use PlatformIO as the build system. The project has a dependency on ArduinoJson for handling JSON documents, and on ArduinoUnit for some tests I have. Currently, I can't add tests for the code that uses ArduinoJson because of my PlatformIO's native environment (*tester*) failing to build when using ArduinoJson as a dependency.

Do I need to add some compile flag?  In #548, setting a compile flag was suggested to someone with a similar setup, but that specific flag doesn't make sense to add in my case. Is there a flag to use in these cases?

I noticed that Arduino is selected in the `library.json`, does it mean that this framework needs to be available in my environment for the library to build successfully?

This is my project's `platformio.ini` file:
```ini
[platformio]
env_default = mega, tester

[env:tester]
platform = native
lib_deps =
  ArduinoJson
  ArduinoUnit
build_flags = -std=gnu++11
src_filter = +<*> -<core/>

[env:mega]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_deps =
  ArduinoJson
  ArduinoUnit
lib_compat_mode = strict
build_flags = -std=gnu++11
src_filter = +<*> -<tester/>
```

And this is the error I get:
```bash
# ...

Archiving .pioenvs/tester/libc0a/libArduinoJson.a
ar: no archive members specified
usage:  ar -d [-TLsv] archive file ...
ar -m [-TLsv] archive file ...
ar -m [-abiTLsv] position archive file ...
ar -p [-TLsv] archive [file ...]
ar -q [-cTLsv] archive file ...
ar -r [-cuTLsv] archive file ...
ar -r [-abciuTLsv] position archive file ...
ar -t [-TLsv] archive [file ...]
ar -x [-ouTLsv] archive [file ...]
*** [.pioenvs/tester/libc0a/libArduinoJson.a] Error 1
================================== [SUMMARY] ==================================
Environment mega       	[SUCCESS]
Environment tester     	[ERROR]
========================== [ERROR] Took 4.21 seconds ==========================
```

Thanks in advance for any help.

bblanchon, 2018-05-06:
Hi @adolfogc,

I have very limited knowledge of PlatformIO, but I think I understand the issue.
ArduinoJson is a header-only library, which means there is no `.cpp` file to compile and, therefore, no `libArduinoJson.a` to produce.

I looked at the PlatformIO documentation, and it seems that we can disable the production of `libArduinoJson.a` with the field [`libArchive`](http://docs.platformio.org/en/latest/librarymanager/config.html#libarchive):

> `libArchive`
>
> Create an archive (*.a, static library) from the object files and link it into a firmware (program). This is default behavior of PlatformIO Build System ("libArchive": true).
>
> Setting "libArchive": false will instruct PIO Build System to link object files directly (in-line). This could be useful if you need to override weak symbols defined in framework or other libraries.

Please try to add `libArchive: false` in the `library.json` of ArduinoJson to see if it fixes the problem.

Regards,
Benoit

adolfogc, 2018-05-06:
Hi @bblanchon,

I tried using ArduinoJson as a private library (downloaded from GitHub) and setting the `libArchive` option in its `library.json` file:
```json

{
  "name": "ArduinoJson",
  "keywords": "json, rest, http, web",
  "description": "An elegant and efficient JSON library for embedded systems",
  "homepage": "https://arduinojson.org/?utm_source=meta&utm_medium=library.json",
  "repository": {
    "type": "git",
    "url": "https://github.com/bblanchon/ArduinoJson.git"
  },
  "version": "5.13.1.testLibArchive",
  "authors": {
    "name": "Benoit Blanchon",
    "url": "https://blog.benoitblanchon.fr"
  },
  "exclude": [
    "fuzzing",
    "scripts",
    "test",
    "third-party"
  ],
  "libArchive": false,
  "frameworks": "arduino",
  "platforms": "*"
}
```

That didn't work, so I tried setting it for my environment using the `lib_archive` option:

```ini
[platformio]
env_default = mega, tester

[env:mega]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_deps =
  ArduinoJson
  ArduinoUnit
lib_compat_mode = strict
build_flags = -std=gnu++11
src_filter = +<*> -<tester/>

[env:tester]
platform = native
lib_deps =
  ArduinoJson       ; <-------- USING PLATFORMIO'S LIB
  ArduinoUnit
build_flags = -std=gnu++11
src_filter = +<*> -<core/>
lib_archive = false ; <-------- CHANGE HERE
```

It worked this time!

Thanks for your help solving this issue, and, thanks for this great library!

closed 2018-05-06 by adolfogc