bblanchon/ArduinoJson #733: Clone from bag of tricks gives an error

emilekm2142, 2018-05-10:
```
JsonVariant clone(JsonBuffer& jb, JsonVariant prototype)
{
	if (prototype.is<JsonObject>()) {
		const JsonObject& protoObj = prototype;
		JsonObject& newObj = jb.createObject();
		for (const auto& kvp : protoObj) {
			newObj[clone(jb, kvp.key)] = clone(jb, kvp.value);
		}
		return newObj;
	}

	if (prototype.is<JsonArray>()) {
		const JsonArray& protoArr = prototype;
		JsonArray& newArr = jb.createArray();
		for (const auto& elem : protoArr) {
			newArr.add(clone(jb, elem));
		}
		return newArr;
	}

	if (prototype.is<char*>()) {
		return jb.strdup(prototype.as<char*>());
	}

	return prototype;
}
```
gives compilation error 
```
JsonObject.hpp: 296:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const ArduinoJson::JsonVariant&, void>
       if (Internals*: StringTraits<TStringRef>::equals(key, it->key)) break
```

pointing this line 
`newObj[clone(jb, kvp.key)] = clone(jb, kvp.value);`

bblanchon, 2018-05-11:
Hi @emilekm2142,

Thank you very much for reporting this bug.
I fixed [the code on the website](https://arduinojson.org/doc/tricks/?utm_source=github&utm_medium=issues).

Regards,
Benoit

bblanchon, 2018-05-12:
Unfortunately, this `clone()` function uses [`JsonBuffer::strdup()`](https://arduinojson.org/api/jsonbuffer/strdup/?utm_source=github&utm_medium=issues) which is deprecated; so, you'll see a warning in the logs.

BTW, why do you need to clone an object?

emilekm2142, 2018-05-14:
When I receive a JSON with "type":"update" I have to edit the last received JSON. Therefore I need to store the last message somehow.

bblanchon, 2018-05-15:
My advice: store the original JSON document because it's much smaller than the equivalent `JsonObject`.

closed 2018-05-23 by bblanchon