bblanchon/ArduinoJson #734: Adding a element to array and save on file

raesmen, 2018-05-10:
Hi everyone.

I am trying to add a new element to a array following [this example/test](https://github.com/bblanchon/ArduinoJson/blob/dd025f7804771842e14ea12543ffcf23f495c4a7/test/Issue289.cpp). Concretely, this is my code:

```
void savePreset(String _name){
  File file = SD.open(presetPositonFile);
  StaticJsonBuffer<512> jsonBuffer;

  JsonObject &root = jsonBuffer.parseObject(file);
  JsonObject& presets = root["presets"].asArray().createNestedObject();
  presets["name"] = _name;
  presets["pos_x"] = positions[POS_X];
  presets["pos_y"] = positions[POS_Y];

  if (root.printTo(file) == 0) {
    Serial.println(F("Failed to write to file"));
  }
  file.close();
}
```
The problem is that I am getting always _Failed to write to file_. I am using Arduino Uno with its own IDE and the compilation spits this warning:

```
warning: 'ArduinoJson::JsonArray& ArduinoJson::Internals::JsonVariantCasts<TImpl>::asArray() const [with TImpl = ArduinoJson::Internals::JsonObjectSubscript<const char*>]' is deprecated (declared at /Users/macbookpro/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp:18): use as<JsonArray>() instead [-Wdeprecated-declarations]
   JsonObject& presets = root["presets"].asArray().createNestedObject();
```

bblanchon, 2018-05-11:
Hi @raesmen,

The warning is pretty clear; it says:

> `asArray()` is deprecated: use `as<JsonArray>()` instead

Concerning the SD file, the program cannot write to it because it's opened as read-only.
To open the file in read-write mode, you need to pass `FILE_WRITE` to `open()`.

You can find a complete example that reads and writes a configuration file to the SD card in the chapter "Case Studies" of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

Regards,
Benoit

raesmen, 2018-05-17:
The content of the file is:

```
{"presets":[{"name":"pos1","pos_x":100,"pos_y":590},{"name":"pos2","pos_x":1000,"pos_y":592},{"name":"pos3","pos_x":1001,"pos_y":5921}]}
```

and after an attempt of the saving function:

```
{"presets":[{"name":"pos1","pos_x":100,"pos_y":590},{"name":"pos2","pos_x":1000,"pos_y":592},{"name":"pos3","pos_x":1001,"pos_y":5921}]}{}
```

It adds a new `{}` at the end of the file.

bblanchon, 2018-05-19:
Hi @raesmen,

It's hard to tell why parsing fails here, but the other problem is obvious: `printTo()` appends to the end of the file, but you expect to replace the content of the file.
It's probably better to close the file after reading it and then create a new one.
You can either rename the old one to keep a backup or simply delete the file.

Regards,
Benoit

raesmen, 2018-05-24:
Hi again @bblanchon 

We are doing what you say, and it works but no 100%.
We have discovered that if run the function with 2 elements on the array the third is written good, but after this, the next objects I try to save don´t go fine.

We will make more more tests, but... could it be related to anything about the size??
This is the new code:

```
void savePreset(String _name){
  File file = SD.open(presetPositonFile);
  DynamicJsonBuffer jsonBuffer;
  JsonObject &root = jsonBuffer.parseObject(file);
  file.close();
  SD.remove(presetPositonFile);

  file = SD.open(presetPositonFile, FILE_WRITE);
  JsonObject& presets = root["presets"].as<JsonArray>().createNestedObject();
  presets["name"] = _name;
  presets["pos_x"] = positions[POS_X];
  presets["pos_y"] = positions[POS_Y];

  int arraySize =  root["presets"].size();
  for (int i = 0; i< arraySize; i++){
    Serial.println(root["presets"][i]["name"].as<String>());
  }

  if (root.printTo(file) == 0) {
    Serial.println(F("Failed to write to file"));
  } else {
    Serial.println(F("Saving a preset success..."));
  }
  
  file.close();
}
```

The for statement is a check point to see what is happening, and the maximum quantity of iterations I can see are three.

PS: I apologize the delay (again), it is a personal project and we meet on Thursdays.

bblanchon, 2018-05-25:
Hi @raesmen,

That could be a memory usage problem.
You can check if [JsonArray::createNestedObject()](https://arduinojson.org/api/jsonarray/createnestedarray/?utm_source=github&utm_medium=issues) successfully allocated the memory by looking at [JsonArray::success()](https://arduinojson.org/api/jsonarray/success/?utm_source=github&utm_medium=issues).

If your platform is very limited in term of RAM (for example, an ATmega), I strongly recommend that you avoid any form of dynamic memory allocation.
In particular, you should replace all `String`s with good old `char[]` and replace the `DynamicJsonBuffer` with a `StaticJsonBuffer`.

Regards,
Benoit

closed 2018-06-12 by bblanchon