bblanchon/ArduinoJson #737: arduinojson decode

Arash-HR, 2018-05-13:
How to convert json to this format :
{\"ip\":\"192.168.1.1\",\"macaddress\":\"ab:3c:5r:95}

RayyanTahir, 2018-05-13:
Hi, I haven't been able to use much of ArduinoJson library and my answer may be wrong, but I think the actual JSON format is like: `{"ip":"192.168.1.1","macaddress":"ab:3c:5r:95"}`

The probably reason you have believed in the format you mentioned in your question is because your console prints or shows the json string that way. the `\` character is generally used to escape the `"` character. Which means that `\"` simply translates to `"`. I hope that answers your confusion.

Arash-HR, 2018-05-13:
@RayyanTahir To use the arduino library, I must have a character entry, so I must change the json template to use this library. Anyway, thank you so much

Arash-HR, 2018-05-13:
Duplicate of #737
To use the arduino library, I must have a character entry, so I must change the json template to use this library. Anyway, thank you so much

Arash-HR, 2018-05-13:
To use the arduino library, I must have a character entry, so I must change the json template to use this library. Anyway, thank you so much .

closed 2018-05-13 by Arash-HR