bblanchon/ArduinoJson #739: Parsing reliably fails after update from Arduino IDE

brownjd, 2018-05-18:
This code:

DynamicJsonBuffer jsonBuffer(getJSONBufferSize());
JsonArray &jsonChart = jsonBuffer.parseArray(f);
    
Serial.print(F("Free heap after chart: "));
Serial.println(ESP.getFreeHeap());
yield();
Serial.println(jsonChart.success())

where f is a SPIFFS file handle to this data stream:

[{"average":271.709},{"average":271.777},{"average":271.62},{"average":271.668},{"average":271.723},{"average":271.779},{"average":271.749},{"average":271.822},{"average":271.767},{"average":271.925},{"average":271.83},{"average":271.627},{"average":271.614},{"average":271.542},{"average":271.502},{"average":271.513},{"average":271.5},{"average":271.379},{"average":271.321},{"average":271.507},{"average":271.36},{"average":271.487},{"average":271.509},{"average":271.608},{"average":271.55},{"average":271.807},{"average":271.882},{"average":271.964},{"average":271.843},{"average":271.82},{"average":271.711},{"average":271.447},{"average":271.265},{"average":271.137},{"average":271.191},{"average":271.006}]

The code reliably fail to parse the JSON stream after updating the ESP8266 NodeMCU with with Arduino IDE. Pressing the reset button does not help. However, if I power cycle the ESP after updating code, it works.

When it works, you'll get this:

updateChartInfo()...
Chart file size: 1330
Free heap before JSON allocation: 32976
Free heap after chart: 14984
updateChartInfo()...done

Why would the behavior be different after a warm reset? Is there any why to get a reason code as why parsing fails. 
 
Using ArduinoJson v5.13.1 with Arduino IDE 1.8.5

Here's the compile output.

Archiving built core (caching) in: /var/folders/bh/1t5dnbwd20q1d6c436jnt_880000gn/T/arduino_cache_880854/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_5ef2a8d9edefc2baf658fcff45032440.a
Sketch uses 418392 bytes (40%) of program storage space. Maximum is 1044464 bytes.
Global variables use 40076 bytes (48%) of dynamic memory, leaving 41844 bytes for local variables. Maximum is 81920 bytes.
warning: serialport_set_baudrate: baud rate 921600 may not work
Uploading 422544 bytes from /var/folders/bh/1t5dnbwd20q1d6c436jnt_880000gn/T/arduino_build_137547/ESPQuoteMachine.ino.bin to flash at 0x00000000
................................................................................ [ 19% ]
................................................................................ [ 38% ]
................................................................................ [ 58% ]
................................................................................ [ 77% ]
................................................................................ [ 96% ]
.............                                                                    [ 100% ]

bblanchon, 2018-05-19:
Hi,

That looks like an undefined behavior, can you share a bit of code?
(max 20 lines of code please)

As an aside, the following JSON document is probably not the best way to store an array of floats:

```json
[{"average":271.709},{"average":271.777},{"average":271.62},{"average":271.668},{"average":271.723},{"average":271.779},{"average":271.749},{"average":271.822},{"average":271.767},{"average":271.925},{"average":271.83},{"average":271.627},{"average":271.614},{"average":271.542},{"average":271.502},{"average":271.513},{"average":271.5},{"average":271.379},{"average":271.321},{"average":271.507},{"average":271.36},{"average":271.487},{"average":271.509},{"average":271.608},{"average":271.55},{"average":271.807},{"average":271.882},{"average":271.964},{"average":271.843},{"average":271.82},{"average":271.711},{"average":271.447},{"average":271.265},{"average":271.137},{"average":271.191},{"average":271.006}]
```

instead, you could simplify to 

```json
{"average":[271.709,271.777,271.62,271.668,271.723,271.779,271.749,271.822,271.767,271.925,271.83,271.627,271.614,271.542,271.502,271.513,271.5,271.379,271.321,271.507,271.36,271.487,271.509,271.608,271.55,271.807,271.882,271.964,271.843,271.82,271.711,271.447,271.265,271.137,271.191,271.006]}
```

This change divides the size of the `JsonBuffer` in half.

To reduce the size even further, you can parse the document in chunks, as explained in [Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/?utm_source=github&utm_medium=issues)

Regards,
Benoit

brownjd, 2018-05-19:
I’m retrieving that JSON from a 3rd party API, so I have no control over it.

I have the code up on github - ESPStockTicker if you want to take a look. That’s probably easiest. I only use the dynamic buffers in two places.

https://github.com/brownjd/ESPStockTicker

Jason Brown

bblanchon, 2018-05-21:
Hi Jason,

When looking at `ESPQuoteMachine.ino`, this function really surprised me:

```c++
int getJSONBufferSize()
{
  Serial.print(F("Free heap before JSON allocation: "));
  Serial.println(ESP.getFreeHeap());
  return ESP.getFreeHeap() - WORKING_HEAP;
}
```

So, it seems that the program looks at the total remaining memory to compute the capacity of the `JsonBuffer`.
I admit that, in theory, it should work, but I'm pretty sure in practice it doesn't.
Even if it does work, it's very brittle because any fragmentation of the heap would prevent the allocation of the `JsonBuffer`.

My advice is: don't try to be too smart. Use the [ArduinoJson Assistant](https://arduinojson.org/assistant/?utm_source=github&utm_medium=issues) to compute the required capacity.
If it's too big, parse the JSON document in chunks, as suggested in [Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/?utm_source=github&utm_medium=issues).

Regards,
Benoit

brownjd, 2018-05-21:
Thanks for looking. Two issues I’m trying to work around: 1) the json gets longer as the day progresses 2) I retrieve it with https so I have to save it to a file to free memory for the dynamic buffer.

I’ll take another look at parsing it in chunks. I did consider that, but I don’t remember why I chose not to.

Jason Brown

brownjd, 2018-05-21:
Is there a way to preprocess the stream to calculate the buffer size required?

Jason Brown

brownjd, 2018-05-21:
I tried chunking it and that seems to be working, so I think you are correct about memory fragmentation.

Thanks for the help!

Jason Brown

bblanchon, 2018-05-22:
You're welcome 😄 
Thanks for the update ~~Json~~Jason.

closed 2018-05-22 by bblanchon