bblanchon/ArduinoJson #740: Float parsing fails on Arduino but failure is position dependent within JSON string

mkarlesky, 2018-05-18:
We're seeing an odd problem parsing floats with version 5.13.1 of the library. After lots of testing it seems to be specific to an AVR build and also seems to be dependent on the order of fields in the JSON string being parsed.

We're compiling ArduinoJson and our own code directly with the avr-gcc toolchain (no Arduino IDE) for use on a mega2560. In our AVR build we define the symbols `ARDUINO` and `ARDUINO_ARCH_AVR`.

JSON message A
`{"Value":27.756, "SensorId": 10, "Location": "North", "Timestamp": "Thu Jan 01 00:00:00.0 1970"}`

JSON message B
`{"SensorId": 10, "Value":27.756, "Location": "North", "Timestamp": "Thu Jan 01 00:00:00.0 1970"}`

When our code parses [A] on an Arduino mega2560 the resulting Value is 0.00. However, when our code parses [B] the parsed Value is 27.756 as expected. Prior to updating the ArduinoJson library to 5.13.1 from 4.9.4 float parsing in these scenarios was working without issue (except for the decimal point problem we submitted a couple weeks ago).

If we exercise the same JSON messages and same code cross-compiled with gcc on Linux in our unit test frameworks, both messages A and B parse as expected.

```
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(json);

_sensorId = root["SensorId"];
_value = root["Value"]; // Sometimes 0.00, sometimes correct value
```
In reality, our code is more complicated than above. We have a base class that handles validating JSON against a schema and child classes derived from it that then do the parsing using ArduinoJson. We reuse DynamicJsonBuffer and clear it before each new parsing run. I can provide the two classes if you care to see them as well as our test code. I'm omitting them only because it's quite a bit to include. At their core the ArduinoJson code is as simple as shown above.

bblanchon, 2018-05-19:
Hi @mkarlesky,

Assuming that you're not using the zero-copy mode (for example, you read the input from a Stream), this looks like a memory fragmentation issue.
I strongly recommend specifying the capacity of the `DynamicJsonBuffer` to the constructor; this way, the `JsonBuffer` will make only one big allocation.

Even better, I suggest using a `StaticJsonBuffer` instead.
As a general rule, I always use a `StaticJsonBuffer` on AVR because the stack size is not limited.
However, since your using another toolchain, it may be different.

Regards,
Benoit

mkarlesky, 2018-05-21:
We tried both DynamicJsonBuffer with capacity specified in its constructor and also a StaticJsonBuffer. In both cases the allocation size (200 bytes) was twice the size of the JSON string being parsed (100 characters).

In both instances we saw the very same behavior described at the beginning of this thread. If "Value" is the first item in the JSON string its float value is parsed as 0.0. If it's the second item then its value is parsed correctly.

Further ideas?

mkarlesky, 2018-05-21:
Very happy to provide anything further. But I need a bit of guidance on what that should be.

We know the problem is specific to our embedded setup. The code only fails when running on our mega2560's (with `ARDUINO` and `ARDUINO_ARCH_AVR` build flags set). The exact same code (minus the build flags) cross-compiled on Linux and exercised with Google Test-based unit tests operates correctly. Further, this problem only presented itself after updating from the 4.9.4 version of ArduinoJson to 5.13.1.

From the preceding, we know the problem is specific to our embedded setup and that something in the difference between 4.9.4 and 5.13.1 has exposed the problem.

An issue like this feels like a byte packing or data boundary issue. But, since 5.13.1 is entirely contained in header files it's compiled with the same settings as the rest of our code. And, we have `ARDUINO_ARCH_AVR` set. So we're not seeing an obvious byte packing or boundary issue.

What further tests can we run or information can we provide?

Thank you for your support in figuring this out.

bblanchon, 2018-05-22:
Hi Mike,

> it's virtually guaranteed to be compiled with the same settings as the rest of our code

True, except if you `#define` different values for `ARDUINO_JSON_XXX` values in different compilation unit.

As I already said, all signs point to a memory fragmentation problem: the `malloc()` fails, so the parsing fails. I'm really surprised that the `StaticJsonBuffer` didn't fix the issue.

You mentioned that you reuse the `JsonBuffer`, which I strongly recommend against.
Are you sure that it's correctly cleared after each iteration?
Can you create a fresh `JsonBuffer` before parsing?

What does `JsonObject::success()` says?
I assume it's false, but it would be very curious if it was not.

I'm running out of ideas.
Do you think you can share some code?

Regards,
Benoit

mkarlesky, 2018-05-23:
We've tried every permutation of `DynamicJsonBuffer` and `StaticJsonBuffer` we can think of — reusing & clearing, not reusing, specifying capacity, not specifying capacity. In all variations, the problem persists.

**But, it appears we've made some progress!** We've narrowed in on the issue. The problem is related to accessing `JsonVariant` or `JsonObject` outside of the method in which it was assigned when said access occurs on a mega2560. Specifically, we've tried a method that returns `JsonVariant`, returns `JsonObject*`, or makes use of an out parameter of `JsonVariant&`. If we modify our code so parsing results do not leave the scope of a method then the problem goes away.

Attached is the code that's been so problematic. Unpackager is a base class that validates JSON against a schema using another library and then parses the JSON using ArduinoJson. ReadingUnpackager is derived from this base class and handles the individual value extractions. Our unit tests of ReadingUnpackager are also included. Everything works great on a desktop system but mysteriously fails on a mega2560.

At very bottom is a code snippet you can paste into the [ArduinoJson Wandbox](https://wandbox.org/permlink/ffzfIp7p1H4eZxZR). This code executes successfully as expected in Wandbox. Variations of this approach in our project also execute successfully _but only within unit tests run in Linux (see attached code)_. Result access from outside the originating method's scope causes incorrect value extraction when run on a mega2560.

Is our approach to extracting parsing results from a method dangerous? But we're getting away with it on desktop class systems? Have we misunderstood our ability to perform this kind of access?

Immediately below are excerpts of the release build compilation steps for the attached source files. Anything funny in here that would cause weird behavior for ArduinoJson? 

---

`avr-g++ -o build/atlas/core/Unpackager.o -c -std=c++11 -mmcu=atmega2560 -Os -flto -fno-exceptions -ffunction-sections -fdata-sections -fshort-enums -DARDUINO -DARDUINO_ARCH_AVR -DF_CPU=16000000UL …`

`avr-g++ -o build/atlas/core/ReadingPackager.o -c -std=c++11 -mmcu=atmega2560 -Os -flto -fno-exceptions -ffunction-sections -fdata-sections -fshort-enums -DARDUINO -DARDUINO_ARCH_AVR -DF_CPU=16000000UL …`

---

```
#include <iostream>
#include "ArduinoJson.h"

// In our code jsonBuffer is an instance variable and not a method parameter
// We've found the out parameter of JsonVariant& to be a key part of the problem
bool parse(const char* json, DynamicJsonBuffer& jsonBuffer, JsonVariant& var) {
    JsonObject& root = jsonBuffer.parseObject(json);
    var = root;
    return var.success();
}

int main()
{
  DynamicJsonBuffer  jsonBuffer(200);

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  JsonVariant root;

  // Test if parsing succeeds.
  if (!parse(json, jsonBuffer, root)) {
    std::cerr << "parse() failed" << std::endl;
    return 1;
  }

  // Fetch values.
  const char* sensor = root["sensor"];
  long time = root["time"];
  double latitude = root["data"][0];
  double longitude = root["data"][1];

  // Print values.
  std::cout << sensor << std::endl;
  std::cout << time << std::endl;
  std::cout << latitude << std::endl;
  std::cout << longitude << std::endl;
    
  return 0;
}
```
---
[code.zip](https://github.com/bblanchon/ArduinoJson/files/2029483/code.zip)

bblanchon, 2018-05-23:
Hi Mike,

>  We've narrowed in on the issue. The problem is related to accessing JsonVariant or JsonObject outside of the method in which it was assigned when said access occurs on a mega2560.

Do you mean that `parseObject()` always returns `true`?

I've looked at the attached zip file, and I cannot spot a specific line where there is a clear error.
However, I think there is room for simplification and improvement.

First, the destructor of `IReadingUnpackager`, as well as every abstract class, must be made `virtual`.

Second, the `JsonBuffer` is used during a very short time (basically only during the execution of `ReadingUnpackager::unpackage()`) but its lifetime is much longer. This is a huge waste of memory.

There are many other things to say, but here is the place for a code review.

I'm sorry Mike, I've tried to help you as much as I could, but I cannot afford to spend more time on this issue unless you can trace the problem back to ArduinoJson.

Regards,
Benoit

mkarlesky, 2018-05-23:
Thank you for your help. We'll keep investigating.

To answer your question, yes, the equivalent of `parseObject()` when executed on our mega2560s returns `true` but value extraction from `JsonVariant` or `JsonObject` is not reliable. On desktop and in Wandbox everything is reliable and works as expected.

As to your recommendations:
* ReadingUnpackager is instantiated once and used for the lifetime of the application. All references to it are always good. Hence, the omission of the destructor. However, you are right that it's more correct to include it.
* Because ReadingUnpackager is used continually we elected to leave JsonBuffer in memory rather than repeatedly remove it and instantiate it again. As we've added a large amount of RAM to our board we have this luxury.

If we find an issue with ArduinoJson we'll submit a separate issue.

mkarlesky, 2018-05-24:
We switched to hunting for the cause of the problem in Valgrind memory profiling logs generated by our unit tests. We found and fixed the problem.

The issue had to do with a memory handling change of some months ago in a supporting class that affected the JSON string being parsed. Moving from version 4.9.4 to 5.13.1 of ArduinoJson exposed the issue. Desktop memory management was tolerating the bad behavior and allowing our locally compiled unit tests to execute successfully. However, on the target embedded platform we weren't so lucky.

While debugging the problem and examining Valgrind logs we came across a few opportunities to add null pointer guards before `strcmp()` calls. We don't know how valuable these may be, but I created a PR for your review. These guards _might_ help with more graceful parsing failures in the future for cases of bad memory.

Thank you again for the great library and your assistance.