bblanchon/ArduinoJson #741: JsonObject suddenly lost from memory

ClemRz, 2018-05-19:
Hi, first of all, thank you so much for this awesome library!

I'm observing a strange behaviour with this code:
```
void writeBackup(JsonObject& root) {
  if (!_sdInitialized) {
    Serial.println(F("SD initialization failed"));
    return;
  }
  if (SD.exists(BACKUP_FILE_PATH)) SD.remove(BACKUP_FILE_PATH);

root.printTo(Serial); Serial.println();

  File file = SD.open(BACKUP_FILE_PATH, FILE_WRITE);

root.printTo(Serial); Serial.println();

  if (!file) {
    Serial.print(F("Failed to create file ")); Serial.println(BACKUP_FILE_PATH);
    return;
  }
  if (root.printTo(file) == 0) {
    Serial.println(F("Failed to write to file"));
  } else {
    Serial.println(F("File saved"));
  }
  file.close();
}
```

I get this output in the console, see the second call to `root.printTo(Serial)`? it got reset and I don't know why...
```
{"status":"success","data":{"nextCheckIn":"5","relays":[1,1,1,0,1,1,1,1,0,0,1,1,0,1,0,0]}}
{}
File saved
```

Software config: Arduino IDE 1.8.4, Ethernet lib. 1.1.2, SD lib. 1.2.2, SPI lib. 1.0.0, ArduinoJson lib. 5.13.1
Hardware config: Arduino Mega + official Ethernet shield.

Thanks for your help.

bblanchon, 2018-05-21:
Hi Clément,

So, from what I understand, it stops working as soon as you had another variable on the stack (`file` in this case).
That sounds like a stack overflow.
Try to reduce the number and size of variables used in the program.
In particular, eliminate every global variable and avoid `String`s.

If you're looking for an in-depth explanation, have a look at the "Missing C++ course" and the "Troubleshooting" chapters of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

Regards,
Benoit

ClemRz, 2018-05-21:
Thanks. I think I'll try with https://github.com/henrikekblad/ArduinoJson, seems lighter.

closed 2018-05-21 by ClemRz

bblanchon, 2018-05-22:
It's the same library from 4 years ago.
Sure, it's lighter in terms of features, but the footprint on the stack is not that different.