bblanchon/ArduinoJson #742: Iterate trough parsed JSON an array

steevesf, 2018-05-20:
Using ArduinoJson on Particle.io, I get a JSON from a web server to be able to know when a relay is tuned on/off with duration and know also if today is a day enable to use the relay. 

I am able to use a JsonArray& to loop trough my relays and print the relay no. 

What I try to make is
- Loop in "days" node and print key + the value
- Loop in "time" node and print the key + the value ex: 09:03 as key ,10 as value

Notes: Relays can be dynamic from 1 to 16 relays. Also the node "time" can be dynamic size

Here is the sample of my json

`{
  "updated": "2018-05-20 00:08:19",
  "relays": [
    {
      "days": {
        "1": "1",
        "2": "1",
        "3": "1",
        "4": "1",
        "5": "1",
        "6": "1",
        "7": "1"
      },
      "time": {
        "06:05": "1",
        "00:17": "30",
        "09:03": "10"
      },
      "no": "1"
    },
    {
      "days": {
        "1": "0",
        "2": "0",
        "3": "1",
        "4": "0",
        "5": "0",
        "6": "1",
        "7": "0"
      },
      "time": {
        "12:30": "2",
        "00:46": "1",
        "00:17": "30",
        "09:03": "5"
      },
      "no": "2"
    }
  ]
}`

or see here https://jsoneditoronline.org/?id=d8ff004d6c774c608ba123c62b4667d5

Thanks for your help

bblanchon, 2018-05-21:
Hi @steevesf,

Please try:

```c++
JsonArray& relays = root["relays"];
for (JsonObject& relay: relays) {
  JsonObject& days = relay["days"];
  for(JsonPair& p : days) {
   const char* key = p.key;
   int value = p.value;
  }

  JsonObject& times : relay["time"];
  for(JsonPair& t : times) {
    const char* key = p.key;
    int value = p.value; 
  }
}
```

Regards,
Benoit.

steevesf, 2018-05-21:
Merci @bblanchon , j'ai fait quelques ajustements et ça fonctionne. Merci 

JsonArray& relays = root["relays"];

 for (JsonObject& relay: relays) {

 Serial.println("Relay #:" + String(relay["no"].asString())); // ok print le no relay

  for (JsonObject& relay : relays) {
    JsonObject& days = relay["days"];
    for(JsonPair& p : days) {
     const char* key = p.key;
     const char* value = p.value;
     Serial.println("Day : " + String(key) + " " + String(value));
    }
    JsonObject& times = relay["time"];
    for(JsonPair& t : times) {
      const char* key = t.key;
      const char* value = t.value;
      Serial.println("Time : " + String(key) + " " + String(value));
    }
  }
 }

closed 2018-05-21 by steevesf

bblanchon, 2018-05-22:
Hi @steevesf,

Cool!

BTW, `asString()` and `asArray()` have been deprecated for ages.
Why do you use them instead of the new `as<T>()` syntax?

Regards
Benoit

steevesf, 2018-05-23:
ok thanks @bblanchon