bblanchon/ArduinoJson #743: Parsing [10,[20,30,40]]

sedwebdev, 2018-05-23:
Hello! I've already read the bag of tricks and so one. Tried a lot but...

Well, that's what I want:

StaticJsonBuffer<30> jsonBuffer;
JsonArray& json = jsonBuffer.parseArray(string, 2);

int first_value = json.get<int>(0); // this gets the 10, it works
int *other_values
int other_values_count = json.get(1).as<JsonArray>().copyTo(other_values); // this doesnt work
// other_values should be = [20,30,40], an array of integers

Thanks

sedwebdev, 2018-05-24:
Oh, also noticed that I was wrong. first_value is not 10 it seems, it's empty.
Also tried json.get<int>(0), not working as well...

bblanchon, 2018-05-25:
Hi @sedwebdev,

Here is the program generated by the [ArduinoJson Assistant](https://arduinojson.org/assistant/?utm_source=github&utm_medium=issues):

```c++
const size_t bufferSize = JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + 20;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "[10,[20,30,40]]";

JsonArray& root = jsonBuffer.parseArray(json);

int root_0 = root[0]; // 10

JsonArray& root_1 = root[1];
int root_10 = root_1[0]; // 20
int root_11 = root_1[1]; // 30
int root_12 = root_1[2]; // 40
```

It says that `[10,[20,30,40]]` is 60 bytes on AVR, so `30` is not enough, even for the zero-copy mode.

Here is how you can use [JsonArray::copyTo()](https://arduinojson.org/api/jsonarray/copyto/?utm_source=github&utm_medium=issues):

```c++
int other_values[10]; // constant to be defined
int other_values_count = root_1.copyTo(other_values);
```

Regards,
Benoit

mircobabini, 2018-05-27:
Thank you very much. Totally missed the assistant!

Also, is it  possible to check if the json string is a valid json?
Or maybe to handle any parse error, if it fails.

Thanks again.

bblanchon, 2018-05-27:
Hi @mircobabini,

The only way to validate a JSON document is to parse it and check the result.

Regards
Benoit

closed 2018-06-04 by bblanchon