bblanchon/ArduinoJson #746: Error while parsing, buffer size calculation issue?

NicolasBonduel, 2018-05-26:
Hey guys, 

I'm trying to parse a JSON object I get from an API. It seems however that I have some issues with calculating the buffer size.
I get the response as a `String` with the `ArduinoHttpClient`, so I first convert it to a `char*` with the ` c_str` function.

Here is how I do it:
```c
  const String response = "{\"one\":true,\"two\":false,\"three\":false}";
  const char* json = response.c_str();
  const size_t bufferSize = JSON_OBJECT_SIZE(3) + 40;
  DynamicJsonBuffer jsonBuffer(bufferSize);
  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println(F("Parsing failed!"));
    return;
  }
  
  bool one = root["one"]; // true
  Serial.println(one);
```
`Serial output: "1"`

This works great! However when I try with a bigger JSON Object, it always fails to parse.
Here is an example:

```c
  const String response = "{\"one\":true,\"two\":false,\"three\":false,\"four\":{\"1\":[255,255,255],\"2\":[255,255,255],\"3\":[255,255,255],\"4\":[255,255,255],\"5\":[255,255,255],\"6\":[255,255,255],\"7\":[255,255,255],\"8\":[255,255,255],\"9\":[255,255,255],\"10\":[255,255,255]}}";
  const char* json = response.c_str();
  const size_t bufferSize = 10*JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(10) + 200;
  DynamicJsonBuffer jsonBuffer(bufferSize);
  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println(F("Parsing failed!"));
    return;
  }
  
  bool one = root["one"]; // true
  Serial.println(one);
```
`Serial output: "Parsing failed!"`

Note that I'm using your assistant to calculate the buffer size (and thank you for that!). I've also been reading this page: https://arduinojson.org/faq/why-parsing-fails/
I know that this issue is probably not directly related to your library as the examples from the assistant are working, but I thought you could give me a hand on this one.

I'm using an arduino UNO with an old Ethernet shield.

bblanchon, 2018-05-26:
Hi Nicolas,

This JSON document is quite big, and this microcontroller has only 2KB of RAM, so you must be extremely careful not to waste any byte.

Please have a look at [How to reduce memory usage?](https://arduinojson.org/faq/how-to-reduce-memory-usage/?utm_source=github&utm_medium=issues), I updated the page for you.
In short, you should replace `String` with `char` arrays and `DynamicJsonBuffer` with `StaticJsonBuffer`.

Regards
Benoit

NicolasBonduel, 2018-05-26:
Hey Benoit,

Thanks for the fast answer, I've been trying to apply everything I could to reduce memory usage.
It's vastly copied from your example so I kept your licence.
I also replaced the `DynamicJsonBuffer` to a `StaticJsonBuffer`.

Here is the full code I ended up with:

```c
// ArduinoJson - arduinojson.org
// Copyright Benoit Blanchon 2014-2018
// MIT License

#include <ArduinoJson.h>
#include <Ethernet.h>
#include <SPI.h>

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial) continue;

  // Initialize Ethernet library
  byte mac[] = {0x91, 0xA4, 0xDF, 0x0D, 0xF0, 0x97};
  if (!Ethernet.begin(mac)) {
    Serial.println(F("Failed to configure Ethernet"));
    return;
  }
  delay(1000);

  Serial.println(F("Connecting..."));

  // Connect to HTTP server
  EthernetClient client;
  client.setTimeout(10000);
  if (!client.connect("api.MYWEBSITE.org", 80)) {
    Serial.println(F("Connection failed"));
    return;
  }

  Serial.println(F("Connected!"));

  // Send HTTP request
  client.println(F("GET /endpoint HTTP/1.0"));
  client.println(F("Host: MYWEBSITE.org"));
  client.println(F("Connection: close"));
  if (client.println() == 0) {
    Serial.println(F("Failed to send request"));
    return;
  }

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  Serial.println(status);
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response: "));
    Serial.println(status);
    return;
  }

  // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    Serial.println(F("Invalid response"));
    return;
  }

  // Allocate JsonBuffer
  // Use arduinojson.org/assistant to compute the capacity.
  const size_t bufferSize = 10*JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(10) + 200;
  StaticJsonBuffer<bufferSize> jsonBuffer;

  // Parse JSON object
  JsonObject& root = jsonBuffer.parseObject(client);
  if (!root.success()) {
    Serial.println(F("Parsing failed!"));
    return;
  }

  // Extract values
  Serial.println(F("Response:"));
  Serial.println(root["one"].as<char*>());

  // Disconnect
  client.stop();
}

void loop() {
  // not used in this example
}

```

I'm hitting a non-public API so I don't want to reveal the name of the website. You can consider it's returning a JSON object that's similar in size to the one I posted earlier.

bblanchon, 2018-05-27:
Hi Nicolas,

There is no question in your message, so I'm assuming this doesn't work as expected and you want to know why. To answer I need to know what happens: does parsing fail? is the value different than expected?

See:

* [Why parsing fails?](https://arduinojson.org/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)
* [Parsing succeeds but I can't read the values!](https://arduinojson.org/faq/parsing-succeeds-but-i-cant-read-the-values/?utm_source=github&utm_medium=issues)

Regards,
Benoit

NicolasBonduel, 2018-05-27:
My bad.
It fails and prints `"Parsing failed!"`. I would assume it's still a memory issue.

I'm also thinking of creating a middleware that would return a csv or something simpler to parse.

bblanchon, 2018-05-28:
Hi Nicolas,

Assuming you have the right value for `bufferSize`, the parsing should succeed or crash the device.
When a `StaticJsonBuffer` is too big, it usually crashes the program, because it corrupts the stack.

From my experience, most of the time, parsing fails because of a bad input.
Did you ~double~triple-check that the input is valid?

Regards,
Benoit

NicolasBonduel, 2018-05-28:
The input is indeed valid. I ran it against a linter with no issue.

bblanchon, 2018-06-04:
Can you put the JSON document on a public HTTP endpoint?
I have an Ethernet shield, so I can give it a try.

NicolasBonduel, 2018-06-04:
Here it is: https://rawgit.com/NicolasBonduel/166cc41f0bf908f4254c18157dda6832/raw/79c96dac8a50bca5e75ca91414db566013390fcc/api.json

bblanchon, 2018-06-04:
As far as I know, we cannot do HTTPS with the Ethernet shield.
We cannot use rawgit.com because it sends a 301 when we access the resource via HTTP.
Can you find a host that accepts plain HTTP?

NicolasBonduel, 2018-06-04:
Is this one working?
http://api.myjson.com/bins/eao36

bblanchon, 2018-06-05:
I modified a few lines from the example JsonHttpClient and it worked.
Full code: https://gist.github.com/bblanchon/0721bf8c81e67334d69d17ef4c8a3401

closed 2018-06-18 by bblanchon