bblanchon/ArduinoJson #747: Upgrade Features ArduinoJson

laercionit, 2018-05-30:
I use Library EDB, a fast-growing library for large data volume.

https://github.com/jwhiddon/EDB

Unfortunately it generates a binary file, which makes it very difficult to change the data structure, basically we have to re-create the database.

I imagined customizing ArduinoJson for this same type of operation.

Changing an entire object or an entire array would be very useful.

bblanchon, 2018-06-01:
Hi @laercionit,

I'm sorry, I don't understand the question.
It seems that you're comparing ArduinoJson to a database, can you explain?

Regards,
Benoit

laercionit, 2018-06-01:
hello @bblanchon

Of course, it really did look pretty bad the way I wrote it.
First of all congratulations on Lib ArduinoJson, she is fantastic.

I have a project I'm working on and I use EDB as lib to save all system settings. EDB allows with a global object to perform very interesting routines.

I need guidance for using JsonObject as a global object. This will allow me to migrate my EDB code to Json.

See this example, is what I intend to do but using Json:

https://github.com/jwhiddon/EDB/blob/master/examples/EDB_SDCARD_Optimized/EDB_SDCARD_Optimized.ino

bblanchon, 2018-06-04:
Hi @laercionit 

My guidance is: don't use a global `JsonObject`, instead create a struct that mirrors the content of the JSON document.
See example here: [JsonConfigFile.ino](https://arduinojson.org/example/config/?utm_source=github&utm_medium=issues)

Regards,
Benoit

laercionit, 2018-06-05:
Hello bblanchon,

Thank you for your guidance.
Unfortunately I already know this example and it is not a challenge for the project.

What is a way to instantiate the JsonObject as a global.

I do not know if you had the opportunity to study an EDB LIB, but it's fantastic. No matter the size of the file, only the record that what do do update, insert or delete is loaded.

With the json layout below, we have the following possibilities:

Limit the maximum size of the database (DBSize), know the number of records (Count), know the maximum number of records (DBSize / RowSize)

When performing the open database I parse the datafile and load the data into a Struct that will temporarily hold the Header data. This allows for quick operations.

What I am trying to do is to work with Json with a large number of objects inside the data array, but it is not necessary to parse the entire file, just the object that I want to change, include or perform any operation.

struct StructJsonDBHeader {
  unsigned int DBSize = 0;
  unsigned int RowSize = 0;
  unsigned int Count = 0;
} JsonDBHeader;

{
"DBSize": 11923216,
"RowSize": 92,
"Count": 3,
"Data": [
    {
        "Date":  6,
        "Month":  6,
        "Day": 5,
        "Hour":13,
        "Minute": 46,
        "Second": 0,
        "Text": "my log record 1" 
   },
    {
        "Date":  6,
        "Month":  6,
        "Day": 5,
        "Hour":12,
        "Minute": 46,
        "Second": 0,
        "Text": "my log record 1" 
   },
    {
        "Date":  6,
        "Month":  6,
        "Day": 5,
        "Hour":13,
        "Minute": 46,
        "Second": 0,
        "Text": "my log record 2" 
   },
    {
        "Date":  6,
        "Month":  6,
        "Day": 5,
        "Hour":14,
        "Minute": 50,
        "Second": 0,
        "Text": "my log record 3" 
   }
 ]
}

bblanchon, 2018-06-12:
Why don't you store the JSON document (I mean the string, not the `JsonObject`) in a row of EDB?

closed 2018-06-24 by bblanchon