bblanchon/ArduinoJson #749: Are hash characters allowed in a JSON string value?

DTChristie, 2018-06-03:
I'm using your great library to implement some IoT messaging with JSON on MQTT topics. To change the colour of an LED strip, the user supplies something like {"led1":[255,127,0]} where the three values are red, green and blue settings.

I would like to provide the possibility for the user to use a standard CSS HEX string, something like {"led1":"#FFCC34"}, but this JSON fails your parser. If I remove the # character, then the parse goes through. I'm assuming that I can check the type of the returned value to see if it is an array or a string, but this is a bit annoying.

If I look at the JSON standards, there is nothing written about # in string values. Could this be a bug?

bblanchon, 2018-06-04:
Hi @DTChristie,

As far as I know, ArduinoJson has no problem with `#` in the input.
See by yourself on wandbox: https://wandbox.org/permlink/c1X2ABwBmQsmF9Pb

The parsing is probably failing for another reason.
See: [Why parsing fails?](https://arduinojson.org/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)

Regards,
Benoit

DTChristie, 2018-06-04:
Hi Benoît

Thanks for your quick answer. I looked at your list of issues:

- The input is not a valid JSON – this is not true!
- The StaticJsonBuffer is too small – can't believe this, the strings I am looking at are around 40 characters long
- The StaticJsonBuffer is too big (stack overflow) – can't believe this either
- The DynamicJsonBuffer fails to allocate memory – not using dynamic buffer
- You called parseObject() instead of parseArray() – the top level is an object, and I call parseObject
- The nesting limit is too low – there is hardly any nesting.

The fact is that the following got parsed OK:
- {"led1":[12,34,56]}
- {"led1":[]}

But the following did not
- {"led1":"#12A563"} – parse failed
- {"led1":"12A563"} – parse didn't fail, but the result type was not a "string"

Do I need to clear the buffer in some way? I use all the calls within a callback routine.

Do you have any ideas what I might be doing wrong?

Thanks, David

bblanchon, 2018-06-05:
Hi David,

I updated the demo to use a StaticJsonBuffer:
https://wandbox.org/permlink/tf3m6pDeGmp4eaVb
As you can try, it's possible to reproduce the bug by reducing the size of the JsonBuffer of a single byte.
So, my guess is that the JsonBuffer is too small.

Of course, I'm assuming that the JsonBuffer is fresh and that you don't use a global JsonBuffer.
See: [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/faq/why-shouldnt-i-use-a-global-jsonbuffer/?utm_source=github&utm_medium=issues)

Regards,
Benoit

DTChristie, 2018-06-06:
Hi Benôit

To my shame I must tell you that I found the fault: I forgot to escape the double-quotes on the command line (Mac Terminal). Mea culpa mea maxima culpa!

Thanks nevertheless for your help!

closed 2018-06-06 by DTChristie

bblanchon, 2018-06-07:
You're welcome @DTChristie.
Thank you for using ArduinoJson.

bblanchon, 2018-06-07:
PS: If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)