bblanchon/ArduinoJson #750: Is this a null string or some other behaviour?

dpetican, 2018-06-04:
I thought I was doing the right thing but the first code works on an esp8266. The cmd key always exists.  The second code crashes when the resp key is not in the json packet.  What is returned if a key does not exist?  Should I test for the existence of a key before making a comparison if its possible a key might not be present?
```

    // Command: socket response to ws1 with a non-zero client ID
    if (request["cmd"] == "scktresp" && request["clientid"] != 0) {
      unsigned long clientId = request["clientid"];
      request.remove("clientid");
      request.remove("cmd");
      char data[1024+1]; // +1 for the terminator
      request.printTo(data);
      sendDataWs1(clientId, data);
      return;
    }

```
```
    // Command: socket response to ws1 with a non-zero client ID
    if (request["resp"] == "scktresp" && request["clientid"] != 0) {
      unsigned long clientId = request["clientid"];
      request.remove("clientid");
      request.remove("resp");
      char data[1024+1]; // +1 for the terminator
      request.printTo(data);
      sendDataWs1(clientId, data);
      return;
    }

```

bblanchon, 2018-06-05:
Hi @dpetican,

With regards to ArduinoJson, this program looks perfectly fine.
However, allocating 1025 bytes on the stack of an ESP8266 is risky.
Indeed, on this platform, the stack is limited to 4KB.

Try to use a `String` instead of a `char[]`.

Regards,
Benoit

dpetican, 2018-06-13:
Stack really wasn't the problem, but I changed to Strings and DynamicJsonBuffer anyway for other reasons and the problem did not go away.

This solved the problem because sometimes the resp key will be present other times not.  I use the fact that if the first condition is false then the second and subsequent conditions will not be evaluated:

`if (request.containsKey("resp") && request["resp"] == "scktresp" && request["clientid"] != 0)
`

Still, why does request["resp"] == "scktresp"  not evaluate to something safe if the resp key is not present?  Thanks.

bblanchon, 2018-06-13:
Hi @dpetican,

Unless you found a bug, it is safe to compare a `JsonVariant` with a string, even when the value is missing from the `JsonObject`.
Please see this demo: https://wandbox.org/permlink/IyELpqvrX73uhok9

I'm pretty sure something else causes the crash.

Regards,
Benoit

closed 2018-06-24 by bblanchon