bblanchon/ArduinoJson #752: ESP32 issue

chegewara, 2018-06-08:
```
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 40MHz, 4MB (32Mb), 921600, Verbose"

In file included from C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal.h:34:0,

                 from C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:35,

                 from sketch\IoT.ino.cpp:1:

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected unqualified-id before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:229:21: note: in expansion of macro 'fpclassify'

   #define isnan(y) (fpclassify(y) == FP_NAN)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:10:6: note: in expansion of macro 'isnan'

 bool isnan(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:229:21: note: in expansion of macro 'fpclassify'

   #define isnan(y) (fpclassify(y) == FP_NAN)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:10:6: note: in expansion of macro 'isnan'

 bool isnan(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:229:21: note: in expansion of macro 'fpclassify'

   #define isnan(y) (fpclassify(y) == FP_NAN)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:10:6: note: in expansion of macro 'isnan'

 bool isnan(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:229:21: note: in expansion of macro 'fpclassify'

   #define isnan(y) (fpclassify(y) == FP_NAN)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:10:6: note: in expansion of macro 'isnan'

 bool isnan(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected unqualified-id before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:225:21: note: in expansion of macro 'fpclassify'

   #define isinf(y) (fpclassify(y) == FP_INFINITE)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:15:6: note: in expansion of macro 'isinf'

 bool isinf(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:225:21: note: in expansion of macro 'fpclassify'

   #define isinf(y) (fpclassify(y) == FP_INFINITE)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:15:6: note: in expansion of macro 'isinf'

 bool isinf(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:225:21: note: in expansion of macro 'fpclassify'

   #define isinf(y) (fpclassify(y) == FP_INFINITE)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:15:6: note: in expansion of macro 'isinf'

 bool isinf(T x) {

      ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:209:4: error: expected ')' before 'sizeof'

  ((sizeof(__x) == sizeof(float))  ? __fpclassifyf(__x) : \

    ^

C:\Users\admin\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/math.h:225:21: note: in expansion of macro 'fpclassify'

   #define isinf(y) (fpclassify(y) == FP_INFINITE)

                     ^

c:\users\admin\documents\arduino\libraries\arduinojson\src\arduinojson\polyfills\math.hpp:15:6: note: in expansion of macro 'isinf'

 bool isinf(T x) {

      ^

IoT:163: error: 'StaticJsonBuffer' does not name a type

 StaticJsonBuffer<100> JSONBuffer;

 ^

C:\Users\admin\Documents\Arduino\IoT\IoT.ino: In function 'void messageReceived(String&, String&)':

IoT:181: error: 'JSONBuffer' was not declared in this scope

   JsonObject& parsed = JSONBuffer.parseObject(JSONMessage);

                        ^

IoT:181: error: 'JSONMessage' was not declared in this scope

   JsonObject& parsed = JSONBuffer.parseObject(JSONMessage);

                                               ^

IoT:183: error: initializer fails to determine size of 'id'

     char id[] = parsed["id"];

                            ^

IoT:183: error: array must be initialized with a brace-enclosed initializer

IoT:184: error: initializer fails to determine size of 'pin'

     char pin[] = parsed["pin"];

                              ^

IoT:184: error: array must be initialized with a brace-enclosed initializer

IoT:187: error: invalid conversion from 'int' to 'const char*' [-fpermissive]

     client.publish('iotbutton/delete/pin', id);

                                              ^

In file included from C:\Users\admin\Documents\Arduino\IoT\IoT.ino:12:0:

C:\Users\admin\Documents\Arduino\libraries\arduino-mqtt-master\src/MQTTClient.h:231:8: note:   initializing argument 1 of 'bool MQTTClient::publish(const char*, const char*)'

   bool publish(const char topic[], const char payload[]) {

        ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: E:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'StaticJsonBuffer' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
Could anyone help me please?

Marcussacapuces91, 2018-06-09:
Hi,

Since 5.13.2 update in Arduino IDE can't compile having `'StaticJsonBuffer' does not name a type`
Back to 5.13.1: it's ok.

Coding for MKR GSM 1400 (Samd21), code as simple as : 
```
StaticJsonBuffer<1000> jsonBuffer;
const JsonObject& root = jsonBuffer.parseObject(body);

alert1 = Alert(root["limit1"], root["hyst1"]);
alert2 = Alert(root["limit2"], root["hyst2"]);

startTime = root["start"];
stopTime = root["stop"];

```

Marcussacapuces91, 2018-06-09:
Why closing ? The issue still exists

Marcussacapuces91, 2018-06-09:
Seems to be som troubles with namespaces : 
* namespace Internals
* namespace ArduinoJson

But not only, I also have compiling errors like : 
```
xxx\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/./../Numbers/../Polyfills/math.hpp:10:6: error: expected unqualified-id before 'sizeof'
 bool isnan(T x) {
      ^
```
@chegewara rename the issue with something like "major regressions in 5.13.2"
Sorry for you @bblanchon (encore du travail :-) )
Thanks anyway for the very usefull lib.

bblanchon, 2018-06-10:
Sorry about that, I'll fix this ASAP.
Meanwhile, please use 5.13.1.

Marcussacapuces91, 2018-06-10:
Hi,

In fact I digged deeper in that point, it seem’s 5.13.2 is fine, just 6.0.0 beta have it !
I do not understand wy 6.0.0 beta is published in ArduinoIDE, as it is a beta.

Regards,

Marc

--
Marc Sibert
marc@sibert.fr

De : Benoît Blanchon
Envoyé le :dimanche 10 juin 2018 14:42
À : bblanchon/ArduinoJson
Cc : Marc Sibert; Comment
Objet :Re: [bblanchon/ArduinoJson] ESP32 issue (#752)

Sorry about that, I'll fix this ASAP.
Meanwhile, please use 5.13.1.
—

Garanti sans virus. www.avg.com

bblanchon, 2018-06-11:
This bug is fixed in [version 6.0.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.0.1-beta)

closed 2018-06-11 by bblanchon

lauron, 2018-06-21:
I still get this error with version 6.0.1

bblanchon, 2018-06-22:
@lauron What is the target platform?
Can you copy/paste the complete error message?