bblanchon/ArduinoJson #754: Can I use ArduinoJson to send a POST to Slack ?

marcio2704, 2018-06-10:
Hi @bblanchon 

Thanks for sharing your knowledge !

I am challenged to send a message to a Slack Channel using my Arduino UNO + Ethernet Shield.  All the Slack WebHook is set up and it works fine from my desktop. Now I am looking for a way to send a Json object to a given https URL... 

For testing purposes, I created this: https://hooks.slack.com/services/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

And a valid Json payload is:

{"text":"my message"}

I looked at your samples and I could not find any specific to send a POST.  I tried to change the JsonBeaconUDP  to send to a Ethernet Client instead, but it doesn´t seem to be working fine.

I would appreciate any help.

Thanks,

bblanchon, 2018-06-11:
Hi @marcio2704 

As far as I know, it's not possible to do HTTPS with Arduino UNO and Ethernet shield.
So, either Slack allows plain HTTP, or you need to use another microcontroller, like an ESP8266.

Concerning the example, doing a POST is very similar to doing a GET, except that you send the JSON document in the request; so I suggest that you start from [JsonHttpClient.ino](https://arduinojson.org/example/http-client/?utm_source=github&utm_medium=issues) instead of [JsonUdpBeacon.ino](https://arduinojson.org/example/udp-beacon/?utm_source=github&utm_medium=issues)

You can find a complete example in chapter 4 of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues) where I show how to perform a POST to send datapoint to Adafruit IO.

Please keep me informed of your progress.

Regards,
Benoit

marcio2704, 2018-06-15:
Hi Benoit,

Thanks for the time dedicated... 

I am not quite sure HTTP vs HTTPS is the issue... and why:

1) Slack is accepting the call... The http 400 error is a server response, so the HTTPs is flowing OK.  If HTTPS was the issue, I would get a kind of handshake error.

2) I also did a test with HTTP using http://webhook.site/d7e144b8-f002-432d-809b-cffbc0725d1a,  which a cool test site btw.  Also didn´t see anything coming in.

Anyway, thanks for your attention

marcio2704, 2018-06-15:
Never mind... you´re right !  I did a test from my desktop using HTTP to Slack and I got the same HTTP 400 Invalid Payload error.

Thanks again !

bblanchon, 2018-06-15:
You're welcome.
I'm sorry it didn't work out.

closed 2018-06-15 by bblanchon