bblanchon/ArduinoJson #756: 5.13.2 broke JsonBuffer

teemue, 2018-06-14:
After updating from 5.13.1 to 5.13.2 neither StaticJsonBuffer nor DynamicJsonBuffer work.
Compiler gives error StaticJsonBuffer/DynamicJsonBuffer' was not declared in this scope. Downgrading back to 5.13.1 and using the same code compiles.

bblanchon, 2018-06-15:
Hi @teemue,

I'm pretty sure everything is okay with 5.13.2.
However, version 6 replaces `JsonBuffer` with `JsonDocument`.

Are you sure that it was 5.13.2, not 6.0?

Regards
Benoit

tobozo, 2018-06-15:
Got the same error after upgrading the library from the library manager in the Arduino IDE.

```error: 'StaticJsonBuffer' was not declared in this scope```

Rolling back to the previous version solved it.

tobozo, 2018-06-17:
@bblanchon is there any way to detect the library version and have a chance to use `JsonDocument` instead of `JsonBuffer` while maintaining a single codebase in external projects that use your library?

tobozo, 2018-06-18:
@bblanchon thanks for that, I'll patch accordingly 👍 

More info on this `error: 'StaticJsonBuffer' was not declared in this scope` [here](https://github.com/tobozo/M5Stack-SD-Updater/blob/b37b6ef88031f81e63dec411e82f8a55f13d07d1/examples/M5Stack-SD-Menu/M5Stack-SD-Menu.ino#L149)

BTW this library can be pulled from the Arduino Library Manager

bblanchon, 2018-06-19:
@tobozo, any idea why version 6 was installed?
Was it a human mistake or did the Arduino Library Manager install the wrong one?

tobozo, 2018-06-19:
@bblanchon every time you tag a release, it appears in the Arduino Library Manager, which prompts the user into updating without specifying it's beta.

```
version - version of the library. Version should be semver compliant. 
1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid
```

suggestion: keep on doing beta on the relative branch but don't tag your beta releases

vdeconinck, 2018-06-20:
Just wanted to add I got bitten too by the automatic update to 6 beta. 
I was getting the error "DynamicJsonBuffer does not name a type".
The cause is obviously the same (and a rollback restored functionality) but the message is somewhat different, so pasting it here to help people seraching for it.

bblanchon, 2018-06-21:
Thanks, @vdeconinck.
I added this message to the [FAQ](https://arduinojson.org/v5/faq/error-jsonbuffer-was-not-declared-in-this-scope/?utm_source=github&utm_medium=issues), so it should be easier to find.

christophs70, 2018-07-14:
Today, got the same error after upgrading the library from the library manager in the Arduino IDE to 6.2.0-beta.

Rolling back to the previous version 5.13.2 solved it.

tobozo, 2018-07-15:
@christophs70 you can implement both syntaxes in your code

```
#if ARDUINOJSON_VERSION_MAJOR==6
  // use new syntax
#else
  // use old syntax
#endif
```

... and never worry about your IDE upgrading the library, see [this example](https://github.com/tobozo/M5Stack-SD-Updater/commit/7610cb1ea17d05434d24723886f9b51721adfb69)

@bblanchon I'm not good enough in C++ to know the drawbacks or even the feasability, but still I'm wondering if some kind of compatibility layer with the previous version is possible.

If StaticJsonBuffer existed and would resolve to StaticJsonDocument, the message `error: 'StaticJsonBuffer' was not declared in this scope` would go away.

If parseObject() would statically call deserializeJson() + as<JsonObject>() (and ignore the DeserializationError error) that would fix the second compilation error while keeping the initial behaviour.

Or maybe that would just open a pandora box :-)

bblanchon, 2018-07-16:
@tobozo If that was possible I would have done it, but unfortunately, there are many fundamental differences between 6 and 5.

The example in the link is not working anymore, you need to remove the references and `success()` is now `isNull()`; so you need to wrap the whole function in the `#if`.
Don't worry, I don't plan on doing any other big breaking change.

fluentart, 2018-07-20:
This caught me off-guard.  The problem is the Arduino IDE now defaults to v6 beta, while the documentation defaults to v5. If its not possible to make the IDE default to v5 instead, it would be better to make the online documentation default to v6, or atleast make it more clear what is going on from the front page.

bblanchon, 2018-07-26:
@fluentart, I added the following warning on the documentation:

[image]

tangoTH, 2018-07-26:
I tried the entire range of 6.x ones until I found this thread.    5.13.2 on Arduino 1.8.5 on OSX compiled

TonJimFit, 2018-09-19:
@bblanchon , Sorry sir. I have a problem about "JsonBuffer was not declared in this scope" then i follow your solution (Downgrade ArduinoJson to 5.x (recommended as v6 is still in beta))

[image]

But version is still 6
Serial.print("ARDUINOJSON_VERSION_MAJOR = ");
   Serial.println(ARDUINOJSON_VERSION_MAJOR);
[image]

so could you please give me an advice.

sorry for my poor english skill.
Best regard

bblanchon, 2018-09-19:
@TonJimFit, there is another `ArduinoJson.h` on your computer.

electron1979, 2018-11-11:
5.13.3 compiles nicely on Arduino IDE 1.8.7.
The Library Manager installed 6.5.0-beta!
Thank you ;) ;)

misha-marinenko-official, 2018-12-16:
Just Use version 9.5.0 this helped for me.

misha-marinenko-official, 2018-12-16:
Its not corrent but it works!

svanscho, 2018-12-20:
For some reason Arduino updates to 6-beta even when you select the 5.x version in the lib manager. Reverting back to 5 fixed the issue for me!

electron1979, 2018-12-20:
It didn't for me. 
It will update to the latest if you press "Update".

LiamGomes27, 2019-01-15:
Guys the code is still not compiling for me even after I downgrade the version on arduinojson library. Can anyone help me out?

LiamGomes27, 2019-01-15:
/*
 Version 0.3 - March 06 2018
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

int device_1 = 2;
int device_2 = 0;
int device_3 = 12;
int device_4 = 4;
int device_5 = 13;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "66c7262d-e0ca-4441-bd17-e87e9fe628aa" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "RTGOMES" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "9846680691" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5c3d91921834ae3082fc3966") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, HIGH);
  } 
  else if (deviceId == "5c3d91b41834ae3082fc3968") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, HIGH);
  }
    else if (deviceId == "5c3d91ca1834ae3082fc396a") // Device ID of third device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, HIGH);
  }
    else if (deviceId == "5c3d91de1834ae3082fc396c") // Device ID of fourth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, HIGH);
  }
    else if (deviceId == "5c3d920f1834ae3082fc396e") // Device ID of fifth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_5, HIGH);
  }

  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
[... 155 lines cut ...]
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 

void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

bblanchon, 2019-01-15:
@LiamGomes27 Please open a new issue.

bblanchon, 2019-02-26:
ArduinoJson 6 finally got out of beta; it's time to close this issue.

closed 2019-02-26 by bblanchon