bblanchon/ArduinoJson #757: Floatingpoint Serialization only expses 6 significant mantissa digits

APokorny, 2018-06-19:
With revision d41f7a816593af4d182c05d445f1792cbf4a0421 the serialization of floating point numbers changed. It only writes 6 digits for mantissa when writing 32bit floats. But 32bit floating point numbers have 23 mantissa bits so including the leading and hidden 1 this means 24 bits. 24 divided by ln(10) yields 7.27..... So 7 digits when printed in decimal would be accurate if I am not mistaken. The same can be said about the double serialization routines. 54 significant mantissa bits means 16 decimal bits vs currently 9 decimal digits.

bblanchon, 2018-06-20:
Hi @APokorny 

I spent a lot of time and energy on this topic.
From my experience, adding a 7th digits only gives you the illusion of accuracy.
I decided to limit to 6 digits to avoid trailing .0000001 and .9999999.

Concerning the limitation to 9 digits, it is imposed by the conversion algorithm.
Please read [the article](https://blog.benoitblanchon.fr/lightweight-float-to-string/) that describes the process.

I'm sorry but this is the I can do.
If you need a better conversion, you'll have to write your own and use [RawJson()](https://arduinojson.org/v5/api/utils/rawjson/?utm_source=github&utm_medium=issues)

Regards,
Benoit

aj-ptw, 2018-06-27:
You can always mulitply by 100,000 or something and then divide by the same number on the other end. It's a simple way to boost the number of decimal places.

closed 2018-07-02 by bblanchon