bblanchon/ArduinoJson #758: Is it possible to use this in a similar fashion to a database?

denden4444, 2018-06-19:
Hi 

Currently I am experimenting with the ESP8266 and SPIFFS.
I have a table which I can add records to and delete them. (I have no index key yet for the records)
I would like to read and write/update a file which contains the table data. The file is not very large in size and currently I have each table row and its data separated by commas as a line in a text file. 
For example : 
Motor1, outside,date,time,on
Motor2, outside,date,time,on
..etc

Does someone please have an example of how to achieve this ? I would really appreciate the help or guidance as the examples here don't quite show this.

Thank you in advance 

Den

bblanchon, 2018-06-20:
Hi @denden4444,

No, ArduinoJson only does serialization/deserialization and doesn't aim to be a database.

Curiously, the same question has been asked in #747. See the answers there.

Regards,
Benoit

closed 2018-07-02 by bblanchon

denden4444, 2018-07-05:
Thanks for the reply ... Will try and see how I can get that done.