bblanchon/ArduinoJson #759: Crash on JsonObject.printTo() String

StefanNienhuis, 2018-06-19:
I want to save a JsonObject to SPIFFS on the Wemos D1 Mini, when I try to print the JsonObject to a String, I get a stack trace. This is my code:

`void saveConfiguration() {
  File configurationFile = SPIFFS.open(configurationFileName, "w");
  String configuration;

  configurationRoot.printTo(configuration);

  if (!configurationFile) {
    Serial.println("Config file could not be loaded");
  } else {
    configurationFile.print(configuration);
  }
}`

And this is the stack trace:
https://pastebin.com/vTuCbZx6

Thanks in advance

bblanchon, 2018-06-20:
Hi Stephan,

The first thing you need to do is to use the [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) so you can have a better idea of what is going on.

Then, have a look at [Why does my device crash or reboot?](https://arduinojson.org/v5/faq/why-does-my-device-crash-or-reboot/?utm_source=github&utm_medium=issues).

Finally, I highly recommend the "Troubleshooting" chapter in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues) as it covers a lot of other topics.

Regards
Benoit

StefanNienhuis, 2018-06-20:
Hi,

Thanks for your quick reply, I've downloaded the ESP Exception Decoder, and this is its output:
https://pastebin.com/asQtvLhx

I've looked in to Why does my device crash or reboot? but I didn't find anything. Could you help me further?

Thanks in advance

bblanchon, 2018-06-21:
Hi @StefanNienhuis,

I wonder if this could be a bug in ArduinoJson 6.
Can you share an [MCVE](https://stackoverflow.com/help/mcve)?
BTW, did you try with ArduinoJson 5?

Regards,
Benoit

StefanNienhuis, 2018-06-21:
Hi @bblanchon,

When I made an MCVE the code does work, but the full code has passed over 500 lines. So it's probably a problem in my code and not in your library.

I'll close the issue since it's a problem with my code.

Thanks for your amazing help!

bblanchon, 2018-06-21:
Hi @StefanNienhuis,

In the past, a bug in the allocator of ArduinoJson caused similar alignment errors.
If in doubt, please reopen this thread.

Good luck.

StefanNienhuis, 2018-06-21:
Hi @bblanchon,

After updating to the last version of your library I'm getting a different error:
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads

Do you have any idea what this means?

Thanks

morty9, 2018-06-24:
Hi @StefanNienhuis, Hi @bblanchon,

I have the same issue in my program, when i printTo my object in String my program crash and i've an exception. Did you find a way to resolve it?

Here is y code :
[image]

Thanks for all ! 
Bérangère

bblanchon, 2018-06-24:
Hi @morty9,

Depending on the platform, a `StaticJsonBuffer` of 2KB could cause a stack overflow.
Can you try with a `DynamicJsonBuffer`?

Also, where is `jsonBuffer` declared?
Is it still in scope when `printTo()` is executed?

If it still doesn't work, can you provide an [MCVE](https://stackoverflow.com/help/mcve)?

Regards,
Benoit

bblanchon, 2018-07-03:
@morty9 and @StefanNienhuis any progress?

StefanNienhuis, 2018-07-03:
Hi @bblanchon,

My problem is solved, the problem was that I had a lot of JsonBuffers. So I took a look at your SaveConfiguration example, and using that code it did work.

Thanks for your amazing help!

Regards,
Stefan

bblanchon, 2018-07-04:
You're welcome @StefanNienhuis, thanks for the update.
I'll wait to see how @morty9 is doing before closing this issue.

morty9, 2018-07-04:
Hi @bblanchon 

I think my problem is solved, i change the allocation of the buffer with DynamicJsonBuffer and it seems to work now.

Thanks for your help and sorry for my late response ! 

Regards, 
Bérangère

bblanchon, 2018-07-04:
That's fine @morty9, thanks for the update.

closed 2018-07-04 by bblanchon