bblanchon/ArduinoJson #762: Compile Error

mars000, 2018-06-23:
I am using 5.13.2 on ESP8266 and I get an error that I cannot work out - compile error screenshot below. Effectively I am trying to get length of json Object and copy into an char array and send that info across to another ESP8266 using ESP-NOW.

Code:

```
DynamicJsonBuffer jBuffer;
JsonObject &root = jBuffer.createObject();
root["ID"] = 2;
root["leak"] = true;
root["time"] = "23:00:01";
root["date"] = "03/04/2019";
  
int json_length_a = (int) root.measurePrettyLength() + 1;  // +1 as I with to add a '\0'
char hold[json_length_a];   // **<--- if i change to char hold[240] i.e. use a number compiles fine !**
Serial.print ("json_length = "); Serial.println(json_length_a);
Serial.print ("hold = "); Serial.println(hold);
  
root.prettyPrintTo(hold);    // **<---- compile error here.**
  uint8_t bs1[sizeof(hold)];
  memcpy(bs1, hold, sizeof(hold)); 
  esp_now_send(remoteMac, bs1, sizeof(hold));`

```
[image]

bblanchon, 2018-06-24:
Hi @mars000,

Unfortunately, the library cannot infer the size of a variable length array (VLA).
Or at least, I'm not aware of any trick that would make that possible.

So, you need to pass the size of the VLA as the second parameter of `prettyPrintTo()`:

```c++
root.prettyPrintTo(hold, json_length_a);
```

By the way, this is quite a gymnastic that you're doing here; why don't you use a fixed size array or a `String`?

Regards,
Benoit

mars000, 2018-06-25:
thanks for quick response - your responses are always quick and succinct.

I'm need to use Dynamic as the example case I'm using it for is more complex than the attached above example and content is changing dynamically. But a String as you suggested could work (I think)

Great library btw !

bblanchon, 2018-06-27:
You're welcome @mars000.
Thanks for using ArduinoJson.

closed 2018-06-27 by bblanchon