bblanchon/ArduinoJson #763: Unable to test key `is<String>()`

sticilface, 2018-06-23:
I'm trying to use some templated functions to retrieve values from a JsonObject see below

```cpp
template<typename T>
bool parse(const __FlashStringHelper * key, JsonObject & root, T & dest)
{
	if (root.containsKey(key) && root[key].is<T>()) {
		dest = root.get<T>(key);
		return true; 
	} else {
		return false;
	}
}

```

and i've discovered that you cannot test a Key for being a `String`.  I know you can test for `const char *` instead but as in my first example it has to be templated so i cannot use `const char *` when passing `String` in... 

here is a minimal sketch
```cpp
#include <ArduinoJson.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); 
  DynamicJsonBuffer buffer;
  JsonObject & root = buffer.createObject();
  String myString = "hello"; 
  root["test"] = myString; 

  if (root["test"].is<String>()) {
    Serial.println("root is String"); 
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

I know String is not included in the list of types you can test for using `is<T>()`.  is there a specific reason for this? 

Cheers

bblanchon, 2018-06-24:
Hi @sticilface,

You're right, it's missing, and there is no real reason.
It's been on my secret todo list for a long time; now the world knows about it :-)

Regards,
Benoit

sticilface, 2018-06-25:
Haha. Ooops! 

I had noticed it a while ago but had not connected the dots as to why. 

Look forward to it!

commit 0b3af166 (2018-10-06): Fixed `JsonVariant::is<String>()` (closes #763)
closed 2018-10-06 by bblanchon