bblanchon/ArduinoJson #765: ESP8266HTTPClient Stream Compatability

jtubb, 2018-06-24:
When using the [ESP8266HTTPClient library ](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino) it seems that I can get at the TCP stream by doing the following:

`WiFiClient * stream = http.getStreamPtr();`

However the following throws all kinds of compile error:

```
WiFiClient* stream = http.getStreamPtr();

DynamicJsonBuffer jsonBuffer(200);

JsonObject& root = jsonBuffer.parseObject(stream);
```

the most important of which is probably:

`no matching function for call to 'makeParser(ArduinoJson::Internals::DynamicJsonBufferBase<ArduinoJson::Internals::DefaultAllocator>*, HTTPClient&, uint8_t&)'`

It's not clear to me exactly what stream type your library is expecting? Do I need to get to a lower level in the wificlient?

bblanchon, 2018-06-24:
Hi @jtubb,

[JsonBuffer::parseObject()](https://arduinojson.org/v5/api/jsonbuffer/parseobject/?utm_source=github&utm_medium=issues) expects a `Stream&` (reference), not a `Stream*` (pointer).
So you simply need to call it like that:

    JsonObject& root = jsonBuffer.parseObject(*stream);

Alternatively, you can change `stream` to be a reference, by calling `getStream()` instead of `getStreamPtr()`:

    Stream& stream = http.getStream();

In the Case Studies of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues), there is a complete example that uses ESP8266HTTPClient to read the forecasts from Weather Underground. Also, if you need a refresher on what are references and pointers, please read The Missing C++ Course at the beginning of the book.

Regards,
Benoit

jtubb, 2018-06-25:
Ahhh thanks. I just assumed there would be an overload for a pointer to the stream. It's all working now!

closed 2018-06-25 by jtubb