bblanchon/ArduinoJson #769: How can I insert an object in a json before

tarzan115, 2018-06-27:
I have a json like
```json
{
	"info":[
		{
			"id": 1,
			"url":"abc.xyz"
		},
		{
			"id": 2,
			"url":"abc.xyz"
                },
		{
			"id": 3,
			"url":"abc.xyz"
		}
                // and much more object...
	]
}
```
read from `SPIFFS`
now I want to insert a new object to that json in SPIFFS.
new object
```json
{
   "id":4,
   "url":"abc.xyz"
}
```

after insert
```json
{
	"info":[
		{
			"id": 1,
			"url":"abc.xyz"
		},
		{
			"id": 2,
			"url":"abc.xyz"
                },
		{
			"id": 3,
			"url":"abc.xyz"
		},
                // .........................
                {
                        "id":4,
                        "url":"abc.xyz"
                }
	]
}
```

please help me, how can I do that.
thanks!

bblanchon, 2018-06-27:
Hi @tarzan115,

In the example above, it seems that you want you to add a new object at the end of the array.
You can do that with [JsonArray::createNestedArray()](https://arduinojson.org/v5/api/jsonarray/createnestedobject/?utm_source=github&utm_medium=issues):

```c++
JsonObject& root = jsonbuffer.parseObject(file);
JsonArray& info = root["info"];
JsonObject& newInfo = info.createNestedObject();
newInfo["id"] = 4;
newInfo["url"] = "abc.xyz";
```

But if you need to *insert* the new object at another position (i.e., not at the end), it's not supported by the library.
If that's what you want, I need to convert this into a feature request.

If you need assistance concerning the SPIFFS part, have a look at the example [JsonConfigFile.ino](https://arduinojson.org/v5/example/config/?utm_source=github&utm_medium=issues), it's written for the SD card, but can easily be adapted for SPIFFS.
If you're looking for a complete example using SPIFFS, please read the Case Studies in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues)

Regards,
Benoit

tarzan115, 2018-07-03:
thank you so much. it's work :D

closed 2018-07-03 by tarzan115

bblanchon, 2018-07-03:
You're welcome.
Thank you for using ArduinoJson.