bblanchon/ArduinoJson #771: reuse StaticJsonBuffer - docs clarification

d21d3q, 2018-06-29:
[Docs](https://arduinojson.org/v5/faq/how-to-reuse-a-jsonbuffer/) are a liitle bit misleading about warnign about reusing (global) buffer. 

Lets take web server use case:
```C++
// we have a global buffer
StaticJsonBuffer<500> json_buffer; 

// and dozen of handler functions which are serializing (global?) state of program
void handler() {
  json_buffer.clear();
  JsonObject& root = json_buffer.createObject();
  root["foo"] = "bar";
  root.prettyPrintTo(some_server);
}
```
Documentation states:
> it’s possible to reuse a JsonBuffer thank to the clear() method. However, it’s very risky and can be avoided most of the time.

It shouldn't be considered risky if one knows what is doing.
If your code has multiple handlers like this one, then having one buffer per function is not so memory wise (since we are sure that we are not using two JosnObjects at the same time).

bblanchon, 2018-06-30:
Hi @d21d3q,

The right way to write this function is:

```c++
void handler() {
  StaticJsonBuffer<500> json_buffer; 
  JsonObject& root = json_buffer.createObject();
  root["foo"] = "bar";
  root.prettyPrintTo(some_server);
}
```

because it avoids the global variable.
It's is a good thing because it saves 500 bytes that are now available for the whole execution of the program.

> then having one buffer per function is not so memory wise

This statement is very puzzling... On the contrary, having one buffer per function is the best option to reduce the memory usage.

Regards,
Benoit.

d21d3q, 2018-06-30:
From docs:
```C++
// STEP1: parse input
StaticJsonBuffer<200> jsonBuffer;
JsonObject& inputObject = jsonBuffer.parseObject(inputJson);

/* ...etc... */

// STEP2: generate output
jsonBuffer.clear();
JsonObject& outputObject = jsonBuffer.createObject();
outputObject["hello"] = inputObject["world"];
```
this is dangerous because two `JsonObject`s are pointing to same buffer and further use of  inputObject can lead to error.

but:
```C++
StaticJsonBuffer<500> json_buffer; 

void first_handler() {
  json_buffer.clear();
  JsonObject& root = json_buffer.createObject();
  root["foo"] = "bar";
  root.prettyPrintTo(some_server);
}

void second_handler() {
  json_buffer.clear();
  JsonObject& root = json_buffer.createObject();
  root["foo"] = "bar";
  root.prettyPrintTo(some_server);
}
```
when not using `json_buffer` outside of handlers, scope is limited and I can allocate memory only once (and not multiply it by number of handlers.
> the best option to reduce the memory usage.

Can you elaborate more about it?

bblanchon, 2018-07-04:
Hi @d21d3q 

When you declare a global `StaticJsonBuffer` like that, you reserve 500 bytes for the whole execution of the program; so you cannot use these bytes for anything else.

When you declare a local `StaticJsonBuffer` in each function, you reserve 500 bytes for a short period; you can reuse these bytes as soon as the function exits.

Don't worry about the execution speed: constructing and destructing a `StaticJsonBuffer` uses just a handful of CPU cycles, the same time as `clear()` takes.

If you want to learn more about this topic, I invite you to read "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues). It should give you a good understanding of how memory is managed in a microcontroller. It's a very common source of struggle among Arduino users, that's why this chapter is at the very beginning of the book.

See also:

Regards,
Benoit

closed 2018-07-12 by bblanchon