bblanchon/ArduinoJson #779: CORRUPT HEAP: Bad head at 0x3ffc9a00. Expected 0xabba1234 got 0x00000000

sean-m-miller, 2018-07-10:
I am experiencing a strange memory error when instantiating a wrapper object for ArduinoJson. I am trying to implement a linked-list of custom json objects from the createNestedJson() function. I first thought the problem would be from Strings, but after switching to const char*, the error persists. I also tried replacing the member JsonObject& fields to be JsonVariant's, and switching out a DynamicJsonBuffer for the StaticJsonBuffer. 

An earlier version of this program, without Custom_Object_Node's allocated on the heap (with just the JsonObject& and StaticJsonBuffer<1024> fields) worked fine.

When the constructor for the Data class is called globally or in a setup() function in Arduino IDE, the program hangs, and fails to ever enter the loop(). When another object's constructor (that passes in the newly constructed Data object as a parameter) is called after the Data constructor, I get the error listed in the title, along with:

> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x40086e27  PS      : 0x00060d36  A0      : 0x80086ed8  A1      : 0x3ffc93f0  
A2      : 0x3ffc6118  A3      : 0xbaad5678  A4      : 0x3ffc610c  A5      : 0x00000080  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x3ffc6168  A9      : 0x3ffc6110  
A10     : 0x00000003  A11     : 0x3ffc62c1  A12     : 0x000160fc  A13     : 0x3ffc3930  
A14     : 0x000001b0  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffe9  

It doesn't seem like this is caused by pointer arithmetic. My next guess would be memory fragmentation, but I do not know how to tell if that's the case. I'm running this on an ESP-WROOM-32 if that helps. Any other suggestions?

Data.h:

```
#ifndef Data_h
#define Data_h

#include "Arduino.h"
#include <ArduinoJson.h>

class Data{

    public:

        Data();

        ~Data();

        template<typename T> void add(const char* key_, T t){ // relies on implicit casts
            DATAroot[key_] = t;
        };

        void create_custom_object(const char*);

        template<typename T> int add_to_custom_object(const char* key_, const char* index, T t){
            Custom_Object_Node* it = obj_head;
            while(it->key != key_ && it != obj_tail){
                it = it->next;
            }
            if(it->key == key_){ // found in list
                it->obj[index] = t;
                return 1; //success
            }
            return 0; //key not found
        }
        
        StaticJsonBuffer<1024> jsonBuffer; 

        JsonObject& DATAroot;

    private:

        class Custom_Object_Node{

            public:

            Custom_Object_Node(const char*, JsonObject&);
            
            JsonObject& obj;

            const char* key;

            Custom_Object_Node* next = NULL;

        };

        Custom_Object_Node* obj_head = NULL;

        Custom_Object_Node* obj_tail = NULL;

};

#endif
```

Data.cpp:

```
#include "Arduino.h"
#include "Data.h"

Data::Data() : DATAroot(jsonBuffer.createObject()){
    const char* key = "data";
    create_custom_object(key);
}

Data::~Data(){}

Data::Custom_Object_Node::Custom_Object_Node(const char* key_, JsonObject& root_) : obj(root_.createNestedObject(key_)), key(key_){}

void Data::create_custom_object(const char* key_){
    Custom_Object_Node* new_obj = new Custom_Object_Node(key_, DATAroot);
    if(obj_head == NULL){ // empty list
        obj_head = new_obj;
        obj_tail = new_obj;
    }
    else{
        obj_tail->next = new_obj;
        obj_tail = new_obj;
    }
}
```

Thanks

sean-m-miller, 2018-07-10:
Update: The previous error was happening due to silly errors on my end. However I have hit another bug that appears to be similar in nature.

 I added a destructor to data.cpp shown below.

```
Data::~Data(){
    custom_deleter(obj_head);
}

void Data::custom_deleter(Custom_Object_Node* it){
    if(it == NULL){
        return;
    }
    custom_deleter(it->next);
    delete it;
}
```
When a Data object on the heap gets destroyed (with or without a linked list), the last recursive custom_deleter() call finishes, and I hit the following error:

> assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c", line 261, function: heap_caps_free
abort() was called at PC 0x400d7097 on core 1
Backtrace: 0x4008d950:0x3ffd2c30 0x4008db2b:0x3ffd2c50 0x400d7097:0x3ffd2c70 0x40083eef:0x3ffd2ca0 0x40085da1:0x3ffd2cc0 0x4000bec7:0x3ffd2ce0 0x40114c7d:0x3ffd2d00 0x400d1956:0x3ffd2d20 0x400d15e4:0x3ffd2d40 0x401151fe:0x3ffd2d60
Rebooting...

Following the Backtrace, it breaks on the line where I call delete on the Data object, but I don't get any more information as to why (After that line, it just calls functions in lower level c libraries). 

Backtrace:

> 0x4008d950: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 139
0x4008db2b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 148
0x400d71cf: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40083eef: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 261
0x40085da1: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 42
0x40114db5: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x400d19ce: WiFiPayload::clear_data(Data&) at /home/sean/Arduino/libraries/WiFiPayload/WiFiPayload.cpp line 161 <--- LINE WHERE I CALL DELETE ON DATA OBJECT

It's possible that this is a general Arduino question, but I still have a hunch that the use of JsonBuffer's and JsonObject's on the heap is the problem. Let me know if you've seen anything like this before.

Thanks again

sean-m-miller, 2018-07-10:
All Fixed! Was from even more errors on my end.

closed 2018-07-10 by sean-m-miller

bblanchon, 2018-07-11:
Hi @sean-m-miller,

I'm glad you found the answer.
Sorry, I couldn't help you with that.
Thank you for using ArduinoJson.

Regards,
Benoit