bblanchon/ArduinoJson #780: How to pass JsonObject between functions of 2 separate classes?

vyruz1986, 2018-07-12:
Hi everyone! I've been using ArduinoJson for a couple of years now (Thanks @bblanchon  & contributors!) but find myself stuck with something seemingly simple. I was struggling with how to pass a JsonObject between 2 functions of 2 separate classes, but after reading through the documentation, I think I should take a step back and think about what I'm doing is correct or not. So here goes:

I have a `WebHandlerClass` and a `SlaveHandlerClass`, the `SlaveHandlerClass` sets up a websocket client connection which receives regular json messages from the remote host, depending on what the message is, the `SlaveHandlerClass` should discard the message, or pass it on to the `WebHandlerClass`. Since I need to check the type of message, I need/want to create a `JsonObject` in the `SlaveHandlerClass` in order to use ArduinoJson functionality to validate the json message format and content. So after validating a json message I have a `JsonObject&` which contains the message I need to pass to `WebHandlerClass`. What would be the appropriate way to pass this data to the other class?
* use `PrintTo()` to print it to a `String` and send that string object to `WebHandlerClass`?
* Pass pointer/reference (these still confuse after years of C++/Arduino coding...) to `JsonObject*` to `WebHandlerClass`?
* Something else...?

I put this on gitter and @bblanchon mentioned to create this issue to further discuss, he already mentioned that passing the `JsonObject` from one class to the other will likely be the best solution, but the challenge will be to keep the `DynamicJsonBuffer` alive for this to work correctly.

I attempted to do this, but it's not working, the relevant code is below (just snippets of my bigger project, so likely won't compile. The full source code can found [here](https://github.com/vyruz1986/FlyballETS-Software/tree/feature/SyncStarting)), I'll try to briefly describe the sequence of relevant events:
* `SlaveHandlerClass::init()` initializes the `__jsonRaceDataBuffer` with the `bsRaceData` global variable as size which is big enough to hold a RaceData json object. (calculated with helper tool on ArduinoJson.org)
* `SlaveHandlerClass` establishes websocket connection to remote host, via the callback function `SlaveHandlerClass::_wsEvent()` it's informed when there's new data.
  * In case of `WStype_TEXT`, the function will create a temporary `DynamicJsonBuffer` and `JsonObject&` to validate json format, and check the type of message. If the message is a RaceData (`request.containsKey("RaceData")` then I use the protected class member `_jsonRaceDataBuffer` and `_jsonRaceData` objects to parse the payload from the websocket again, but now it should be kept alive since these objects are protected members, global to the `SlaveHandlerClass
* `WebHandlerClass::_SendRaceData()` function will perform the following tasks:
  * Create a `DynamicJsonBuffer` object with the size of `bsRaceDataArray` (which should be big enough to hold an array of 2 RaceData objects), and initialize a `JsonObject` with a nested `JsonArray`
  * Create another `DynamicJsonBuffer` / `JsonObject` combination to store its own RaceData (`jsonMasterRaceData `)
  * If a slave is connected, create another `DynamicJsonBuffer` / `JsonObject` combination to store the slave data (`jsonSlaveRaceData`)
  * Fill the `jsonMasterRaceData` object with data, and add it to the `jsonRaceData` array object
  * If a slave is present, query the `SlaveHandlerClass` for the latest slave RaceData, and add it to the `jsonRaceData` object as well
  * Send the RaceData array (`jsonRaceData`) object (either with or without slave racedata) out to all connected websocket clients

The problem is I can't figure out how to correctly pass the data to the WebHandlerClass function. The `JsonObject` I get back from calling `SlaveHandlerClass::getSlaveRaceData()` is either empty, or contains rubbish, or the system crashes alltogether.

### [SlaveHandler.h](https://github.com/vyruz1986/FlyballETS-Software/blob/4ff6f069e003ecb65bcc21b081720522432b7f37/SlaveHandler.h)
```cpp
class SlaveHandlerClass
{
protected:
   DynamicJsonBuffer _jsonRaceDataBuffer(bsRaceData); //bsRaceData is a global variable that contains the size for a RaceData json object
   JsonObject& _jsonRaceData;
   WebSocketsClient _wsClient;
   void _WsEvent(WStype_t type, uint8_t * payload, size_t length);
public:
   void init();
   void loop();
   JsonObject* getSlaveRaceData();
}
extern SlaveHandlerClass SlaveHandler;
```
### [SlaveHandler.cpp](https://github.com/vyruz1986/FlyballETS-Software/blob/4ff6f069e003ecb65bcc21b081720522432b7f37/SlaveHandler.cpp)
```cpp
#include "SlaveHandler.h"
#include <WebSocketsClient.h>
#include <ArduinoJson.h>
void SlaveHandlerClass::init()
{
   this->_wsClient.begin(this->_RemoteIP.toString(), 80, "/ws");
   this->_wsClient.onEvent(std::bind(&SlaveHandlerClass::_WsEvent, this
      , std::placeholders::_1
      , std::placeholders::_2
      , std::placeholders::_3));
   DynamicJsonBuffer _jsonRaceDataBuffer(bsRaceData);
}

void SlaveHandlerClass::loop()
{
   this->_wsClient.loop();
}

void SlaveHandlerClass::_WsEvent(WStype_t type, uint8_t * payload, size_t length)
{
   switch (type) {
   case WStype_DISCONNECTED:
      this->_SetDisconnected();
      break;

   case WStype_CONNECTED:
      this->_bConnected = true;
      this->_bWSConnectionStarted = false;
      this->_AnnounceSlaveIfApplicable();
      Serial.printf("[WSc] Connected to url: %s\n", payload);

      // send message to server when Connected
      //this->_wsClient.sendTXT("Connected");
      break;

   case WStype_TEXT:
   {
      Serial.printf("[WSc] got text: %s\n", payload);
      DynamicJsonBuffer jsonBufferRequest;
      JsonObject& request = jsonBufferRequest.parseObject(payload);
      if (!request.success()) {
         syslog.logf_P(LOG_ERR, "Error parsing JSON!");
      }
      if (request.containsKey("RaceData")) {
         JsonObject& _jsonRaceData = _jsonRaceDataBuffer.parseObject(payload);
      }
      else if (request.containsKey("SystemData")) {

      }
      break;
   }

   case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);

      // send data to server
      // webSocket.sendBIN(payload, length);
      break;
   }
}

JsonObject* SlaveHandlerClass::getSlaveRaceData()
{
   return _jsonRaceData;
}
```

vyruz1986, 2018-07-12:
Update: after messing around with this for another couple of hours, and being thrown at with cryptic compiler messages, I decided to give ArduinoJson v6 beta a try, especially since I read this in the [v6 docs](https://arduinojson.org/v6/doc/upgrade/): 

> With ArduinoJson 5, it was very difficult to use a JsonObject or a JsonArray as a class member because you had to make sure that the JsonBuffer stayed in memory too. The trick was to add the JsonBuffer as a class member too, but it was more complex than it should be.

Which sounds like it could help my cause. The latest commit of my code now works about as well as it did with v5, but still passing the string back and forth. I'll working on passing a JsonDocument around tomorrow

bblanchon, 2018-07-13:
Hi @vyruz1986,

I've been pondering the code samples for a while, but I just can't get my head around it.
I don't understand what the code does and I don't understand what you're trying to acheive.

There is a line that is very puzzling:

```c++
DynamicJsonBuffer _jsonRaceDataBuffer(bsRaceData);
```

What is this line?
It's in the class declaration, so at first, I thought it was a function declaration, but it's also in the `SlaveHandlerClass::init()` function, so it's also a variable definition.

Can you shorten the sample? I'm pretty sure that you can summarize the problem in less than 30 lines.

Regards,
Benoit

vyruz1986, 2018-07-14:
So hear's a minimal example with ArduinoJson v6 where I'm trying to accomplish what I need to do in my real project.

`FooClass` gets json from some external source (in the example below it's a fixed string), that json object consists of an array of objects. I need to store the first array element because this needs to be passed to `BarClass`.

`BarClass` Create a new `jsonBarArray` and add its own generated data (`jsonBar`) to the `jsonBarArray` object.
Then `BarClass` calls `FooClass::getLatestJson()` to get a `JsonObject` from foo, which it adds to the `jsonBarArray` object.

Unfortunately this is crashing for me, I've added some serial prints to identify where the crash is occurring, and it seems to be when calling `serializeJson()` (6 can be seen in the serial monitor, 8 not).

### ArduinoJsonClassTest.ino
```cpp
#include "Bar.h"
#include "Foo.h"
void setup()
{
   Serial.begin(115200);
   Foo.init();
   Bar.init();
}

void loop()
{
   Foo.loop();
   Bar.loop();
}
```

### Foo.h
```cpp
#include <ArduinoJson.h>
class FooClass
{
 protected:
    DynamicJsonDocument _FooDoc;
    JsonObject _jsonFoo;
    unsigned long _ulLastUpdate;

 public:
	void init();
   void loop();

   JsonObject getLatestJson();
};

extern FooClass Foo;
```

### Foo.cpp
```cpp
#include "Foo.h"

void FooClass::init()
{
   _ulLastUpdate = 0;
   _jsonFoo = _FooDoc.to<JsonObject>();
}

void FooClass::loop()
{
   if (millis() - _ulLastUpdate > 1000) {
      _ulLastUpdate = millis();
      
      uint32_t ulRand = esp_random();
      char json[] = "[{\"randomNumber\": 3954071004}]";

      DynamicJsonDocument tmpDoc;
      DeserializationError error = deserializeJson(tmpDoc, json);

      if (error) {
         Serial.printf("deserializeJson() failed: %s\r\n", error.c_str());
         return;
      }
      JsonArray tmpJson = tmpDoc.as<JsonArray>();
      
      _jsonFoo = _FooDoc.to<JsonObject>();
      _jsonFoo = tmpJson[0];
   }
}

JsonObject FooClass::getLatestJson()
{
   return _jsonFoo;
}

FooClass Foo;
```

### Bar.h
```cpp
class BarClass
{
 protected:
    unsigned long _ulLastJsonQuery;
 public:
	void init();
   void loop();
};

extern BarClass Bar;
```

### Bar.cpp
```cpp
#include "Bar.h"
#include "Foo.h"
void BarClass::init()
{
   _ulLastJsonQuery = 0;
}

void BarClass::loop()
{
   if (millis() - _ulLastJsonQuery > 1500) {
      _ulLastJsonQuery = millis();

      Serial.println("1");
      //Create array which needs to contain foo and bar objects
      DynamicJsonDocument barArrayDoc;
      JsonArray jsonBarArray = barArrayDoc.to<JsonArray>();

      Serial.println("2");
      //Create local bar object
      DynamicJsonDocument barDoc;
      JsonObject jsonBar = barDoc.to<JsonObject>();
      jsonBar["randomNumber"] = esp_random();

      Serial.println("3");
      //Add bar oject to array
      jsonBarArray.add(jsonBar);
      
      Serial.println("4");
      //Get foo object from Foo class
      DynamicJsonDocument fooDoc;
      JsonObject jsonFoo = fooDoc.to<JsonObject>();
      jsonFoo = Foo.getLatestJson();

      Serial.println("5");
      //Add foo object to array
      jsonBarArray.add(jsonFoo);

      
      String strJsonFoo;
      Serial.println("6");
      serializeJson(jsonBarArray, strJsonFoo);
      Serial.println("8");
      Serial.printf("[Bar] Got json from Foo: %s\r\n", strJsonFoo.c_str());
   }
}

BarClass Bar;
```

bblanchon, 2018-07-14:
Hi @vyruz1986,

First, thank you for such a clear question; I can see that you put a lot of effort to make it understandable despite being long. I wish more people did the same.

The crash seems to be caused by a use-after-free. In the following lines, the programs stores a reference to a temporary:

```c++
DynamicJsonDocument tmpDoc; // create temporary
DeserializationError error = deserializeJson(tmpDoc, json);

JsonArray tmpJson = tmpDoc.as<JsonArray>();
      
_jsonFoo = _FooDoc.to<JsonObject>();
_jsonFoo = tmpJson[0]; // store the reference

// but soon after, the temporary is destructed
```

If I understand correctly, the role of `FooClass` is to parse an array and return the first element.
We can do it, like that:

```c++
#include "Foo.h"

void FooClass::init()
{
   _ulLastUpdate = 0;
}

void FooClass::loop()
{
   if (millis() - _ulLastUpdate > 1000) {
      _ulLastUpdate = millis();
      
      uint32_t ulRand = esp_random();
      char json[] = "[{\"randomNumber\": 3954071004}]";

      DeserializationError error = deserializeJson(_FooDoc, json);

      if (error) {
         Serial.printf("deserializeJson() failed: %s\r\n", error.c_str());
         return;
      }
      
      _jsonFoo = tmpDoc.as<JsonArray>()[0];
   }
}

JsonObject FooClass::getLatestJson()
{
   return _jsonFoo;
}
```

Unfortunately, `_FooDoc` contains the whole array, which may use a lot of memory.
In that case, you should consider parsing only the object after skipping the `[`.
(see [Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/v5/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/?utm_source=github&utm_medium=issues))

I didn't look as closely at the rest of the code, but I saw that the following block:

```c++
DynamicJsonDocument fooDoc;
JsonObject jsonFoo = fooDoc.to<JsonObject>();
jsonFoo = Foo.getLatestJson();
```

can be simplified to:

```c++
JsonObject jsonFoo = Foo.getLatestJson();
```

Regards,
Benoit

vyruz1986, 2018-07-20:
Thanks for your answer @bblanchon , and sorry or taking so long to respond, I only had time today to sit down and take a good look at your provided answer.

Unfortunately your suggested solution does not work because of a requirement that I did not specify clearly enough in my question:
`FooClass` can receive multiple different JSON messages, the example only provided one hardcoded message, but in the real code these are coming in via a websocket client. I will probably have to store 3 different json objects, and discard a couple of ones I don't want.

`BarClass` (in this example) needs to be able to request the latest data for those 3 objects, and this is happening asynchronously from how `FooClass` is receiving the data, so I must store it somehow in `FooClass` until `BarClass` comes along and requests it (in fact I need to store it indefinitely, since `BarClass` will not store it internally and just request it again from `FooClass` everytime it needs it).

That is why I had the `DynamicJsonDocument tmpDoc;` in place, so I could create a temporary location to check which type of object I received, and then copy it to the appropriate class-level document.

Writing this I think I figured out a solution, I just need to define the temporary `DynamicJsonDocument` at the class-level scope as well, same as the other objects. I'll give that a try and report back

vyruz1986, 2018-07-20:
Hmm, I seem to have hit a snag trying what I mentioned in my previous comment...

Foo.cpp now looks like this, all other files remain as they were in [my comment from last week](https://github.com/bblanchon/ArduinoJson/issues/780#issuecomment-404982194).

### Foo.cpp
```cpp
#include "Foo.h"

void FooClass::init()
{
   _ulLastUpdate = 0;
   _jsonFoo = _FooDoc.to<JsonObject>();
   _jsonInput = _InputDoc.to<JsonObject>();
}

void FooClass::loop()
{
   if (millis() - _ulLastUpdate > 1000) {
      _ulLastUpdate = millis();
      
      Serial.println();
      Serial.println();

      uint32_t ulRand = esp_random();
      char json[50];
      sprintf(json, "[{\"test\": %lu}]", ulRand);
      
      DeserializationError error = deserializeJson(_InputDoc, json);

      if (error) {
         Serial.printf("deserializeJson() failed: %s\r\n", error.c_str());
         return;
      }

      _jsonInput = _InputDoc.as<JsonArray>();
      
      String strJsonInput;
      serializeJson(_jsonInput, strJsonInput);
      Serial.printf("[Foo] Input contains: %s\r\n", strJsonInput.c_str());

      _jsonFoo = _FooDoc.to<JsonObject>();
      _jsonFoo = _jsonInput[0];

      String strJsonFoo;
      serializeJson(_jsonFoo, strJsonFoo);
      Serial.printf("[Foo] Foo contains: %s\r\n", strJsonFoo.c_str());
   }
}

JsonObject FooClass::getLatestJson()
{
   String strJsonFoo;
   serializeJson(_jsonFoo, strJsonFoo);
   Serial.printf("[Foo] Returning json from Foo: %s\r\n", strJsonFoo.c_str());
   return _jsonFoo;
}
FooClass Foo;
```

Watching the output in the serial monitor, now produces this:
```
[Foo] Input contains: [{"test":2013919561}]
[Foo] Foo contains: {"test":2013919561}

[Foo] Returning json from Foo: {"�?":2013919561}
```

The `Input contains` line is printed after I deserialize the `char json[50]` string.
The `Foo contains` line is printed after I copied the first object of the array to the `_jsonFoo` object.
So far everything looks fine.

However, when `BarClass` calls `FooClass::getLatestJson()` and `FooClass` returns the `_jsonFoo` object, suddenly the property name of the object is messed up...

bblanchon, 2018-07-21:
Hi @vyruz1986,

By passing a `char[]` to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/?utm_source=github&utm_medium=issues), you triggered the zero-copy mode of ArduinoJson.
In this mode, ArduinoJson stores pointers to the input buffer instead of making copies.

To fix this program, you need to disable the zero-copy mode by passing a read-only input; to do that, simply cast the input to a `const char*`:

```c++
DeserializationError error = deserializeJson(_InputDoc, (const char*)json);
```

Regards,
Benoit

vyruz1986, 2018-07-21:
Thanks @bblanchon , that fixed it, the example code is now working as expected! I'll go and try to apply what I learned to my project now.

I also just bought your book as a way to say thanks, I was thinking of writing up a short tutorial about this question I asked, with my example code to go along with it. Let me know if you're interested, and if so, how I should provide you the article.

vyruz1986, 2018-07-21:
Hello again, 10 hours of messing around further, and I'm still stuck :(

I realized my example was a little over-simplified. and while the solution you provided works for that example, it does not work for my real code.
The reason is, as I stated in my previous comment, that the `_InputDoc` object is re-used for several messages which are coming in via a websocket client, and depending on the content, they need to be discarded, or copied to a correct other `DynamicJsonDocument` object.

I replicated this behaviour in an updated example code, which I'll attach here. I've created 2 'dummy input' functions which are run in `FooClass` at different intervals, they are both using the `_InputDoc` to deserialize a json string, and then copy the `JsonObject` to the corresponding global one.

Then in `BarClass`, I alternatingly call `FooClass::getJsonFoo1()` and `FooClass::getJsonFoo2()` to get the corresponding data.

It seems somehow the `DynamicJsonDocuments` are not copying the data as I was expecting, but rather still pointing to the input object, as the function for getting Foo1 will sometimes return data which should be in Foo2, and vice-versa. (Mostly Foo2 data is being returned because this input is being dummy-generated more often than Foo1).

I get the following output if I run this on my ESP32:
```
[Foo-R2] Returning json from Foo2: {"randomNumberFrom1":307583337}
[Bar] Got json from Foo: {"randomNumberFrom1":307583337}, should contain randomNumberFromFoo2
[Bar] Have array: [{"randomNumber":147939102},{"randomNumberFrom1":307583337}]

[Foo-R1] Returning json from Foo1: {"randomNumberFrom2":3983978232}
[Bar] Got json from Foo: {"randomNumberFrom2":3983978232}, should contain randomNumberFromFoo1
[Bar] Have array: [{"randomNumber":3458691154},{"randomNumberFrom2":3983978232}]
```

Attached example code:
[ArduinoJsonClassTest.zip](https://github.com/bblanchon/ArduinoJson/files/2216461/ArduinoJsonClassTest.zip)

bblanchon, 2018-07-25:
Indeed, ArduinoJson never makes copies of objects.

For example, the next line gets a reference, not a copy.

```c++
_jsonFoo2 = _jsonInput[0];
```

I plan on adding copy from documents to documents in version 6, but I don't know when it will be available. Meanwhile, I'm afraid I have no solution.

However, I can see three ways to workaround this problem:

1. use a `struct` to store your data
2. deserialize in chunks, so you have only one document
3. serialize() and deserialize() to make a copy

Of course, I would go for option 1, because it decouples the program from the JSON library and the JSON schema; it also gives a strong type to the data and is best in term of memory usage.

bblanchon, 2018-09-11:
Starting with [v6.4.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.4.0-beta), arrays and objects are copied.
That should fix this problem.

commit b106b1ed (2018-09-11): Copy JsonArray and JsonObject, instead of storing pointers (fixes #780)
closed 2018-09-11 by bblanchon