bblanchon/ArduinoJson #784: Exception 3 in version 6.2.2

tarzan115, 2018-07-19:
hi @bblanchon 
I got `Exception 3` with version 6.2.2 beta, maybe I am doing something wrong.
here is a small code:
```c
#include <ArduinoJson.h>

const char TEST_1[] PROGMEM = "TEST_1";
const char TEST_2[] PROGMEM = "TEST_2";
const char TEST_3[] PROGMEM = "TEST_3";
const char TEST_4[]                   = "TEST_4";

void genJson()
{
    StaticJsonDocument<200> doc;
    JsonObject root  = doc.to<JsonObject>();

    root[FPSTR(TEST_3)] = FPSTR(TEST_4);
    root[FPSTR(TEST_1)] = FPSTR(TEST_2);
    serializeJson(root, Serial);
    return;
}

void setup()
{
    Serial.begin(115200);
    genJson();
    return;
}

void loop(){}
```

here is decode stack:
```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x402031c8
EXCVADDR: 0x402352e4: tcp_input at core/tcp_in.c line 376

Decoding stack results
0x402352d4: tcp_input at core/tcp_in.c line 853
0x402352e4: tcp_input at core/tcp_in.c line 376
0x402352dc: tcp_input at core/tcp_in.c line 376
0x4020251d: xPortGetFreeHeapSize at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\heap.c line 181
0x40202c95: stopWaveform at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_waveform.c line 212
0x40203428: uart_init at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\uart.c line 487
0x40100584: realloc at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.c line 1513
0x402021d4: __pinMode at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring_digital.c line 50
0x40202132: __pinMode at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring_digital.c line 44
0x402352e4: tcp_input at core/tcp_in.c line 376
0x40202cf6: stopWaveform at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_waveform.c line 222
0x40203050: uart_read_char at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\uart.c line 100
0x401000e5: umm_disconnect_from_free_list at D:\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.c line 1164
```
thank you!

tarzan115, 2018-07-19:
It works with version 6.1.0

bblanchon, 2018-07-19:
[ArduinoJson 6.2.3-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.2.3-beta) fixes this bug.

There is a small error in the sample above: `TEST_4` should not be passed to `FPSTR()` since it's not in `PROGMEM`.

Again, thank you very much @tarzan115.

commit 3f666bd5 (2018-07-19): Fixed exception when using Flash strings as object keys (fixes #784)
closed 2018-07-19 by bblanchon