bblanchon/ArduinoJson #785: Garabage data at first name in JsonObject,for returned JsonObject from class member function

pulkit-jain, 2018-07-20:
Hi all 

JsonObject& GetJsonRoot() {
	File lConfigFile = SPIFFS.open(CONFIG_FILE_PATH, "r");
	DynamicJsonBuffer lJsonBuffer;
	JsonObject* lJsonRoot = NULL;
	if (!lConfigFile) {
		Serial.println("Failed to open config file");
		return *lJsonRoot;
	}

	size_t lFileSize = lConfigFile.size();

	std::unique_ptr<char[]> lFileData(new char[lFileSize]);
	lConfigFile.readBytes(lFileData.get(), lFileSize);

	lConfigFile.close();

	 lJsonRoot = &lJsonBuffer.parseObject(lFileData.get());

	if (!lJsonRoot->success()) {
		Serial.println("Failed to parse config file");
		return *lJsonRoot;
	}
	lJsonRoot->printTo(Serial);
	return *lJsonRoot;
}

Actually when i print lJsonRoot just at the end of GetJsonRoot,it is giving correct data.
But when i print the in the caller method it is giving garbage data for only first name of data.

Correct Data:-{"serverName":"api.example.com","accessToken":"128du9as8du12eoue8da98h123ueh9h98"}
Incorrect Data:- {"⸮":"api.example.com","accessToken":"128du9as8du12eoue8da98h123ueh9h98"}

Thanks in advance for the help.

Kind Regards
Pulkit Jain

pulkit-jain, 2018-07-21:
I am adding code again in clear format.
```
JsonObject& GetJsonRoot() {
	File lConfigFile = SPIFFS.open(CONFIG_FILE_PATH, "r");
	DynamicJsonBuffer lJsonBuffer;
	JsonObject* lJsonRoot = NULL;
	if (!lConfigFile) {
		Serial.println("Failed to open config file");
		return *lJsonRoot;
	}

	size_t lFileSize = lConfigFile.size();

	std::unique_ptr<char[]> lFileData(new char[lFileSize]);
	lConfigFile.readBytes(lFileData.get(), lFileSize);

	lConfigFile.close();

	 lJsonRoot = &lJsonBuffer.parseObject(lFileData.get());

	if (!lJsonRoot->success()) {
		Serial.println("Failed to parse config file");
		return *lJsonRoot;
	}
	lJsonRoot->printTo(Serial);
	return *lJsonRoot;
}
```

bblanchon, 2018-07-21:
Hi @pulkit-jain,

See:

* [FAQ: Why does the generated JSON document contain garbage?](https://arduinojson.org/v5/faq/why-does-the-generated-json-contain-garbage/?utm_source=github&utm_medium=issues) (examples 1 and 3 in particular)
* [Example: JsonConfigFile.ino](https://arduinojson.org/v5/example/config/?utm_source=github&utm_medium=issues) that shows how to avoid the `unique_ptr` gymnastic.

Regards,
Benoit

pulkit-jain, 2018-07-21:
thanks Benoit!
I am able to resolve my problem by your input.

bblanchon, 2018-07-22:
You're welcome @pulkit-jain.
Thank you for using ArduinoJson and good luck with your projects.

closed 2018-07-22 by bblanchon