bblanchon/ArduinoJson #787: IDE wants to update to Version 6 beta

W4KRL, 2018-07-26:
Using the Arduino version 1.8.5 IDE with ArduinoJson 5.13.2. Windows 10 PC. The IDE continually pops up a notification that there are library updates available. It wants to install the latest 6.x beta.

Perhaps it would be better to keep the beta off the Arduino library system until released.

SConaway, 2018-07-27:
Solution: add a .development file to v5's root.

bblanchon, 2018-07-28:
Hi,

I understand that the update notification is annoying, but I don't have a better solution.
Let me explain.

People have been asking for new features, and the only way I could deliver them was through a different API. Breaking changes are a pain in the ass, so I want to bundle them in a big v5 to v6 transition. I want to make sure that all breaking changes are behind before inviting everyone to switch to the new API, hence the "beta" suffix.

I understand that you would prefer that the beta would not be available through the library manager; but if I do that, nobody will provide any feedback on the new version. This feedback is crucial to understand how people use (or misuse) the library to see how I can improve their experience in the long run. I firmly believe in the virtues of "release early / release often" principle in producing the best possible software, and this is what is happening here.

Curiously, the Arduino Library Manager requires library developers to respect semantic versioning (as I do) but ignores it completely when it comes to update notification. Indeed, we could expect that upgrading 5.13.2 to 6.2.3-beta should be flagged as rather risky and therefore should not be suggested as an automatic update. Hopefully, this feature will happen someday because all libraries are bound to evolve and to make some breaking changes too.

That means that for a few months, you'll have to ignore the notification or disable it entirely. Another option is to uninstall ArduinoJson and download the single header distribution instead. This solution allows using different versions for multiple projects.

Thank you for your understanding.

Regards,
Benoit

W4KRL, 2018-07-28:
> I get what you're saying but I don't completely agree. Any user suitable
> for beta testing should have no problem doing a manual installation of this
> library, especially if you provide some instructions in the documentation.
> That would defeat your use of Library Manager to get more beta testers.
> This is something I've been thinking about for a while and had it on my
> "to-do" list to open a feature request. I've now done so:
> arduino/Arduino#7837 <https://github.com/arduino/Arduino/issues/7837>. I
> welcome any feedback on that proposal

Benoit,

Thank you for your quick response. Also, please accept my compliments for
your excellent book "Mastering ArduinoJson".

I understand that you are trying to get feedback for the v6 betas but I
think you are sending a conflicting message. I sell a display kit that uses
ArduinoJson. The buyers of my kit do not know anything about ArduinoJson
and are unlikely to provide you with any feedback. I write the code using
v5.13.2, but I will have to instruct my users to download that version and
ignore the IDE messages to update. I would like to use v6 but you say
the API will change so I will have a big effort to satisfy unhappy
customers if I switch to v6.

I think that you should make developers aware of the v6 betas and ask them
for feedback without bothering casual users who are happy with v5.

Thanks,
Karl Berger

https://w4krl.com/iot-kits/#remote-display_assembly

*73 Karl W4KRL*

bblanchon, 2019-02-26:
ArduinoJson 6 finally got out of beta; it's time to close this issue.

closed 2019-02-26 by bblanchon