bblanchon/ArduinoJson #788: Unable to allocate nested array(s)

michelep, 2018-07-26:
Hi all,

i'm unable to let my code working:

DynamicJsonDocument historyDoc;
...
{
 JsonObject root = historyDoc.as<JsonObject>();

  JsonArray timestamp = root.createNestedArray("ts");
  JsonArray hist_t = root.createNestedArray("t");
  JsonArray hist_h = root.createNestedArray("h");

  Serial.println(timestamp.isNull()); 
  Serial.println(hist_t.isNull()); 
  Serial.println(hist_h.isNull()); 
  
  timestamp.add(NTP.getTime());
  hist_t.add(dht_last_t);
  hist_h.add(dht_last_h);
  if(hist_t.size() > HISTORY_MAX_SIZE) {
    timestamp.remove(0);
    hist_t.remove(0);
    hist_h.remove(0);
  }       
}

isNull() on each of the three nested arrays return me True, that means "unable to allocate the object". How i can solve or debug this ?

Tnx !

michelep, 2018-07-26:
Oh, i forgot: i'm using latest beta v6.2.3

bblanchon, 2018-07-27:
Hi @michelep,

ArduinoJson takes its little footprint and its speed from the monotonic allocator used in `JsonDocument` (formerly in `JsonBuffer`). As a monotonic allocator, it can allocate memory but cannot release it.

As a consequence, `JsonArray::remove()` removes the element from the `JsonArray` but doesn't release the memory in the `JsonDocument`. Which means you cannot use [`JsonArray::remove()`](https://arduinojson.org/v6/api/jsonarray/remove/?utm_source=github&utm_medium=issues) in a loop.
That what is explained in the red box "Causes memory leaks" [in the documentation](https://arduinojson.org/v6/api/jsonarray/remove/?utm_source=github&utm_medium=issues).

Are you really doing serialization/deserialization with this program?
It seems that you use the `JsonDocument` to store the state of the program.

Why don't you use a simple array of struct instead?
You could then serialize the array in one shot, so the `JsonDocument` would have a very short life.
In a sense, that's what I recommend in [What's the best way to use the library?](https://arduinojson.org/v5/faq/whats-the-best-way-to-use-the-library/?utm_source=github&utm_medium=issues).

Regards,
Benoit

closed 2018-08-23 by bblanchon