bblanchon/ArduinoJson #794: I can not get data

mcmega, 2018-08-13:
Hi, BBlanchon!

I need help.
Use ESP8285.

```cpp
void onEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t * data, size_t len) {
  if (type == WS_EVT_DATA) {
    Serial.println((char*)data);        // I get a valid package (as below)
    DynamicJsonDocument doc;
    DeserializationError error = deserializeJson(doc, (char*)data);
    if (error) {
      Serial.println("Error Pack");
      return;
    }
    JsonObject root = doc.as<JsonObject>();

    if (root["setSC"]) {
      Serial.println( root["setSC"]);  // I get nothing???
    }
  }
}
```
This is the received package from websocket:
```json
{"setSC":
  [
    {
      "ID":0,
      "NM":"-= 0 =-",
      "EN":false,
      "AP":"10",
      "EV":[{"da":"82894F","pr":[[0,5],[0,4]]}],
      "AC":[{"da":"82894F","pr":[[3,2],[0,2]]}]
    },
    {
      "ID":1,
      "NM":"-= 1 =-",
      "EN":true,
      "AP":"12",
      "EV":[{"da":"82894F","pr":[[1,3],[0,0]]}],
      "AC":[{"da":"82894F","pr":[[2,3],[0,4]]}]
    }
  ]
}
```
How do I get the whole array of objects?

beicnet, 2018-08-13:
Here is an example for you @mcmega 

```
const size_t bufferSize = 4*JSON_ARRAY_SIZE(1) + 13*JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(1) + 4*JSON_OBJECT_SIZE(2) + 2*JSON_OBJECT_SIZE(6) + 190;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "{\"setSC\":[{\"ID\":0,\"NM\":\"-= 0 =-\",\"EN\":false,\"AP\":\"10\",\"EV\":[{\"da\":\"82894F\",\"pr\":[[0,5],[0,4]]}],\"AC\":[{\"da\":\"82894F\",\"pr\":[[3,2],[0,2]]}]},{\"ID\":1,\"NM\":\"-= 1 =-\",\"EN\":true,\"AP\":\"12\",\"EV\":[{\"da\":\"82894F\",\"pr\":[[1,3],[0,0]]}],\"AC\":[{\"da\":\"82894F\",\"pr\":[[2,3],[0,4]]}]}]}";

JsonObject& root = jsonBuffer.parseObject(json);

JsonObject& setSC0 = root["setSC"][0];
int setSC0_ID = setSC0["ID"]; // 0
const char* setSC0_NM = setSC0["NM"]; // "-= 0 =-"
bool setSC0_EN = setSC0["EN"]; // false
const char* setSC0_AP = setSC0["AP"]; // "10"

const char* setSC0_EV0_da = setSC0["EV"][0]["da"]; // "82894F"

int setSC0_EV0_pr00 = setSC0["EV"][0]["pr"][0][0]; // 0
int setSC0_EV0_pr01 = setSC0["EV"][0]["pr"][0][1]; // 5

int setSC0_EV0_pr10 = setSC0["EV"][0]["pr"][1][0]; // 0
int setSC0_EV0_pr11 = setSC0["EV"][0]["pr"][1][1]; // 4

const char* setSC0_AC0_da = setSC0["AC"][0]["da"]; // "82894F"

int setSC0_AC0_pr00 = setSC0["AC"][0]["pr"][0][0]; // 3
int setSC0_AC0_pr01 = setSC0["AC"][0]["pr"][0][1]; // 2

int setSC0_AC0_pr10 = setSC0["AC"][0]["pr"][1][0]; // 0
int setSC0_AC0_pr11 = setSC0["AC"][0]["pr"][1][1]; // 2

JsonObject& setSC1 = root["setSC"][1];
int setSC1_ID = setSC1["ID"]; // 1
const char* setSC1_NM = setSC1["NM"]; // "-= 1 =-"
bool setSC1_EN = setSC1["EN"]; // true
const char* setSC1_AP = setSC1["AP"]; // "12"

const char* setSC1_EV0_da = setSC1["EV"][0]["da"]; // "82894F"

int setSC1_EV0_pr00 = setSC1["EV"][0]["pr"][0][0]; // 1
int setSC1_EV0_pr01 = setSC1["EV"][0]["pr"][0][1]; // 3

int setSC1_EV0_pr10 = setSC1["EV"][0]["pr"][1][0]; // 0
int setSC1_EV0_pr11 = setSC1["EV"][0]["pr"][1][1]; // 0

const char* setSC1_AC0_da = setSC1["AC"][0]["da"]; // "82894F"

int setSC1_AC0_pr00 = setSC1["AC"][0]["pr"][0][0]; // 2
int setSC1_AC0_pr01 = setSC1["AC"][0]["pr"][0][1]; // 3

int setSC1_AC0_pr10 = setSC1["AC"][0]["pr"][1][0]; // 0
int setSC1_AC0_pr11 = setSC1["AC"][0]["pr"][1][1]; // 4
```

beicnet, 2018-08-14:
Oh, okay @bblanchon didn't know that, I'm still on v5.12.0 and it's fine for me.

mcmega, 2018-08-15:
Well thank you BBlanchon! I will use it as is now.

bblanchon, 2018-08-31:
This bug is fixed in [v6.3.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.3.0-beta)

closed 2018-08-31 by bblanchon

mcmega, 2018-09-03:
Hi, bblanchon!
You have closed the question. How to correctly get the contents of the task above?

mcmega, 2018-09-03:
Yes thank you!
But I need to format the output to UART.
I did this:
```cpp
if (root["setSC"]) {
        String toUART = "";
        serializeJson(root["setSC"], toUART);
        sendUART("SET", "SC", toUART.c_str());
}

void sendUART(const char *header, const char *argument, const char *value) {
  if (value) {
    Serial.printf( "%c%s%c%s%c%c%s%c%02X%c", SOH, header, US, argument, US, STX, value, ETX, crc8(value, strlen(value)), EOT );
  } else {
    Serial.printf( "%c%s%c%s%c%c", SOH, header, US, argument, US, EOT );
  }
}
```