bblanchon/ArduinoJson #795: Memory leak on esp8266-12e

rafik73github, 2018-08-15:
Code:

```
String url = "/api/v7/timers";
 client.print("GET " + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
                "ApiCode: " + apicode + "\r\n" +
                 "Content-Type: application/json\r\n" + 
                  "Connection: close\r\n\r\n");

 // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    
    return;
  }

 // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    
    return;
  }                   
  
// Allocate JsonBuffer
  // Use arduinojson.org/assistant to compute the capacity.
    const size_t capacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
  DynamicJsonBuffer jsonBuffer(capacity);

  // Parse JSON object
  JsonObject& root = jsonBuffer.parseObject(client);
  if (!root.success()) 
  {
  HT.setLedNow(112);
    conn_alert = true;
      return;
}
else
{
 conn_alert = false;
}
Serial.println(capacity);
Serial.println(ESP.getFreeHeap());

runningIndex = root["timerInfo"][0]["runningIndex"]; 

 
  // Disconnect
 
  client.stop();
```

This code is repeated every 2 seconds. The amount of memory gradually decreases to zero. What should I do to make this not happen? Regards.

bblanchon, 2018-08-17:
Hi @rafik73github,

I don't see any memory leak in this code.
Are you sure that the problem comes for here?

Regards,
Benoit

rafik73github, 2018-08-19:
You're right, the code is OK. The reason is the Arduino IDE version. On Linux, I have the Arduino 1.8.5 version installed, and when I program ESP from it, everything is OK. On Windows 10, I have Arduino 1.8.2, and when I use it to program ESP, there is a memory leak problem. Strange, right? ESP tile settings are identical on both versions. Interestingly, the correct version in the serial monitor displays over 28000 bytes memories and does not fall below. On Arduino 1.8.2 it starts over 23000 bytes and falls to zero.