bblanchon/ArduinoJson #796: ESP8266 can not parsing json message from RaspberryPI

anhtien17, 2018-08-15:
I am doing a project which connect esp8266 to raspberrypi via MQTT. However, I can convert json message into an object. It can be seen the topic "piData", and the result of Serial.println((char*)payload). But after parsing, I couldnot get values from the parsed object (Serial.println(actuatorType); don't show anything).
Please help me with this stuck.

[image]

`//=======================================================================
void callback(char* topic, byte* payload, unsigned int length) {
  Serial.println("===========================");
  Serial.print("Message arrived in topic: ");
  Serial.println(topic);

//  char s[length];
//  for (int i = 0; i < length; i++) {
//    //Serial.print((char)payload[i]);
//    s[i] = payload[i];
//  }
  Serial.print((char*)payload);
  StaticJsonBuffer<800> JSONBuffer;                     //Memory pool
  JsonObject& parsed = JSONBuffer.parseObject((char*)payload); //Parse message
  if (!parsed.success()) {   //Check for errors in parsing
    Serial.println("Parsing failed");
    delay(1000);
    return;
  }

  device2 = parsed["Receiver"];           //Get name of receiver
  actuatorType = parsed["ActuatorType"];           //Get actuator type value
  Serial.println(actuatorType);
  numbActuator = parsed["NumbActuator"];

  for (int i = 0; i < numbActuator; i++)
  {
    orderStatus[i] = parsed["OrderStatus"][i];                          //Get value of sensor measurement
  }
  delay(500);
}
`

bblanchon, 2018-08-17:
Hi @anhtien17,

I'm sorry, I don't understand your question.
From what I see, there is no "ActuatorType" in the received object.
Can you explain further?

Regards,
Benoit

anhtien17, 2018-08-17:
Yes. it's my fault in setting wrong the received object. I've already fixed it.
However, Another question has risen which is:
When I send a json message frequently (period = 10s), sometimes there are some unexpected symbols appear in the end of the received json message (i.e. the tail of the json message), and sometimes don't. 
Forexample: It should receive this message: 
"SENSING_DATA" " = {\"Device\": \"Receiver\", \"SensorType\": \"SonarSensor\", \"numbSensor\": 2, \"SensorData\": [10, 20]}"
but in some periods, the message I gotten are:
"SENSING_DATA" " = {\"Device\": \"Receiver\", \"SensorType\": \"SonarSensor\", \"numbSensor\": 2, \"SensorData\": [10, 20]}"0, 20]}"

Could you explain this issue? and any solution for this?
Thank you.

bblanchon, 2018-08-28:
Hi @anhtien17,

Either a null-terminator is missing or the size is not specified when sending the data.
Can you share a **small** piece of code?

Regards,
Benoit

anhtien17, 2018-08-29:
Here is my full code. 
I want to receive json message from Arduino and forward it to Raspberry Pi. However, communication between Arduino <-> ESP has the above error.

`/* Cấu trúc bản tin json "SENSING_DATA" " = {\"Device\": \"Receiver\", \"SensorType\": \"SonarSensor\", \"numbSensor\": 2, \"SensorData\": [10, 20]}"
   Cấu trúc bản tin json "ORDERING" " = {\"Device\": \"Receiver\", \"ActuatorType\": \"OrderLED\", \"numbActuator\": 2, \"OrderStatus\": [0, 1]}"
*/

#include <PubSubClient.h>           //Add library PubSubCLient to run MQTT protocol on ESP
#include <SoftwareSerial.h>         //Add library SoftwareSerial to use digital pins as hardware serial pins
#include <SerialCommand.h>          //Add library SerialCommand to implement serial communication between Arduino and ESP
#include <ESP8266WiFi.h>            //Auto added when select board NodeMCU 0.9 (ESP-12)
#include <ArduinoJson.h>            //Add library ArduinoJson to create and prase a json message

//include library to check free RAM on ESP8266
extern "C" {
#include "user_interface.h"
}

//============================================================================================
// Declare global variables
const char* device1;        // To get name of Arduino in the json mess from Arduino
const char* device2;        // To get name of Arduino in the json mess from Arduino

const char* sensorType;     // To get type of sensor in the json mess from Arduino
const char* actuatorType;   // To get type of actuator in the json mess from RaspPI

int numbSensor;             // To get number of sensor in the json mess from Arduino
int numbActuator;           // To get number of actuator in the json mess from RaspPI

int sensorData[2];          // To get data from sensor in the json mess from Arduino
int orderStatus[2];         // To get data from control in the json mess from RaspPI

//Variables to define transmission period
const unsigned long sendingPeriod = 5000UL;   // Loop after each 5 seconds
unsigned long sendingPeriod_temp = 5000;

//============================================================================================
// Declare a wifi network
//const char* ssid = "IoT-TLU Lab";               //at lab
//const char* password =  "@iottlu010717";

const char* ssid = "AT";                          //at home
const char* password =  "@t170890";

//const char* ssid = "AndroidAP";                 //from Android phone
//const char* password =  "fysc4822";

// create a new instance for wifi communications
WiFiClient espClient;                             
PubSubClient client(espClient);                   

//============================================================================================
// Declare information of broker
//const char* mqttServer = "192.168.43.172";      // IP address from Android phone
//const int mqttPort = 1883;                      // MQTT port

const char* mqttServer = "192.168.1.32";          // IP address at home
const int mqttPort = 1883;                        // MQTT port

const char* mqttUser = "brokerPI";                // Username to connect to broker
const char* mqttPassword = "iotlab";              // Password to connect to broker

//============================================================================================
// Declare Serial Communications between Arduino and ESP8266
const byte RX = D3;       // Use softserial library to make a digital pin D3, D2 to RX, TX pins
const byte TX = D2;

// Create a new instance for serial communications
SoftwareSerial mySerial(RX, TX, false, 256);
SerialCommand sCmd(mySerial); // Khai báo biến sử dụng thư viện Serial Command
[... 173 lines cut ...]

      Serial.print("failed with state ");
      Serial.print(client.state());
      delay(2000);

    }
  }

  //--------------------------------------
  //Subscribe the mqtt topic: piData""
  client.subscribe("piData");

//Setup the instance sCmd to catch the message from ESP and run a callback function recvJsonFromArduino
  sCmd.addCommand("DATA", recvJsonFromArduino);     
  Serial.println("Da san sang nhan lenh");
}

//==============================================================================
//Main loop
void loop()
{
  client.loop();
  // Once it has done all it needs to do for this cycle, go back to checking if we are still connected.
  sCmd.readSerial();
}`

closed 2018-09-17 by bblanchon