bblanchon/ArduinoJson #797: Question:  How to find a nested item below an unknown value?

deejayspinz, 2018-08-15:
Hi,

First off love the library and have bought your e-book.  I'm stumped with an issue I am sure has an easy answer but I can't seem to find a similar challenge after searching high and low.  Below is a paired down sample of a stock response (paired down as the full response is 15k with 100 items).  My issues are twofold, the dataset is too large to parse, so I am looking to find a way to iterate through it.  When I do, my other issue is the value **2018-08-14** (**below**) is dynamic and unknown at the time of parsing.  I want to parse the values below (i.e "1. open" etc).  How can I get this value without knowing the date value?  Is there a way to use a reference identifier instead of a field name?

Specs:  ArduionJSON 5.13.2  Arduino IDE 1.8.5.  NodeMCU ESP8266

{
    "Meta Data": {
        "1. Information": "Daily Prices (open, high, low, close) and Volumes",
        "2. Symbol": "FLT.V",
        "3. Last Refreshed": "2018-08-14",
        "4. Output Size": "Compact",
        "5. Time Zone": "US/Eastern"
    },
    "Time Series (Daily)": {
        "**2018-08-14**": {
            "1. open": "1.5500",
            "2. high": "1.5700",
            "3. low": "1.5000",
            "4. close": "1.5500",
            "5. volume": "138700"
        },
        "2018-08-13": {
            "1. open": "1.5900",
            "2. high": "1.5900",
            "3. low": "1.5200",
            "4. close": "1.5200",
            "5. volume": "113500"
        },
        "2018-08-10": {
            "1. open": "1.5700",
            "2. high": "1.6200",
            "3. low": "1.5600",
            "4. close": "1.5900",
            "5. volume": "101100"
        }
    }
}

bblanchon, 2018-08-17:
Hi @PlastiBots,

Thank you very much for the kind word and for buying the book.

The only solution I know for this problem is to iterate over all the date objects an get the member in the loop:

```c++
JsonObject& dailySeries = root["Time Series (Daily)"];

for (auto p : dailySeries) {
   const char* day = p.key;
   float openValue = p.value["1. open"];
   // etc...
}
```

Is it what you want, or do you have something else on your mind?

Regards,
Benoit

PlastiBots, 2018-08-17:
Thanks.  I'll give that a try as I think its' what I am looking for.  Ideally I'm only looking for the first set of data but will not know the "2018-08-14" value.   I suppose I can just break out of the for loop after the first iteration - that should work.

closed 2018-08-17 by PlastiBots