bblanchon/ArduinoJson #798: "error: expected identifier before numeric constant" when using the pre-allocated size constructor for DynamicJsonBuffer

sean-m-miller, 2018-08-15:
I am using ArduinoJson for a custom library. I am trying to have a DynamicJsonBuffer as a field of a class, but I am getting the compiler error listed in the title.

header file (test.h):

```
#include "Arduino.h"
#include <ArduinoJson.h>

class Data{
    DynamicJsonBuffer jsonBuffer(2000);
};
```

test.ino:

```
#include <test.h>

void setup() {}

void loop() {}
```

From [this](https://github.com/bblanchon/ArduinoJson/issues/164) link, it seems as though DynamicJsonBuffer jsonBuffer(size_t) is valid syntax. What is very strange is that by simply having that function call in a setup() of a .ino (NOT in a header file), the error disappears:

```
#include <ArduinoJson.h>

void setup() {
  DynamicJsonBuffer jsonBuffer(2000);
}

void loop() {}
```

I am using ArduinoJson version 5.13.2 for the esp32 WROOM and using ArduinoIDE version 1.8.5.

Thanks

bblanchon, 2018-08-17:
Hi @sean-m-miller,

This error is not related to ArduinoJson, it's the syntax for calling the constructor that is not respected.

Indeed, you cannot call the constructor where you declare the member.
Instead, you must call it in the class constructor:

```c++
class Data {
public:
  Data() : jsonBuffer(2000) {}
private:
    DynamicJsonBuffer jsonBuffer;
};
```

BTW, there are several caveats to using a `JsonBuffer` as a class member.
In this case, it's easier to use a [DynamicJsonDocument](https://arduinojson.org/v6/api/jsondocument/) from version 6.

Regards,
Benoit

sean-m-miller, 2018-08-23:
Hi Benoit,

Thanks for your help. Are the "caveats" to using a JsonBuffer as a class member documented anywhere? 

I am currently allocating a StaticJsonBuffer on the heap and having a class member point to it. The program runs for a while but then exhibits some strange behavior. Is it possible that manually allocating the StaticJsonBuffer on the heap or having a StaticJsonBuffer* as a class member could cause memory issues?

Unfortunately, switching to DynamicJsonDocuments would take too much time, so I will just try to make work what I have now (if possible).

Thanks

bblanchon, 2018-08-28:
Hi @sean-m-miller,

I don't think there is a document about that, but you'll quickly realize that it requires jumping through many hoops. I designed v6 for this use case.

Switching to a `JsonDocument` should be pretty straightforward; see [the guide on arduinojson.org](https://arduinojson.org/v6/doc/upgrade/?utm_source=github&utm_medium=issues)

Regards,
Benoit