bblanchon/ArduinoJson #799: Error parsing char json string into an object, when the length of the string doesn't match the size of the array?

agathauy, 2018-08-16:
Hello, I've read the FAQ about this but am still having problems with parsing a JSON string. I'm using an Arduino Uno for testing.

Here's a snippet of the situation:

```
  char buffer[240];
  // [Insert other processing stuff to get data into buffer...]

  Serial.print(F("buffer: ")); Serial.println(buffer); // {\"m\":120000,\"s\":1,\"us\":0,\"t\":1534385686}
  Serial.print(F("strlen(buffer): ")); Serial.println(strlen(buffer)); // 48 so this is correct
  Serial.print(F("sizeof(buffer): ")); Serial.println(sizeof(buffer)); // 240

  Serial.print(F("fM()= "));
  Serial.println(freeMemory()); // 18 -- free memory shows just these amount of bytes remaining

  StaticJsonBuffer<200> jsonBuffer2; // I increased it a bit compared to ArduinoJSON assistant    since that worked sometimes for memory issues
  JsonObject& root2 = jsonBuffer2.parseObject(encoded2); //root2 since I parse a HTTP response to   get the above buffer from BASE64 & AES decryption
   if (!root2.success()) {
          Serial.println(F("CP")); // Would reach this so there's an error
	   return -2; // Device error
   }
```

However, if I use the same code and parse this instead of "encoded2" it works.
`
  const char* encoded2 = "{\"m\":120000,\"s\":1,\"us\":0,\"t\":1534356559}";
`

I know that buffer[240] is getting overwritten, but I'm not sure what's causing this wrong behavior, or if it's a bug since I searched the docs for the issue. And I'm not getting any luck looking at the code.

agathauy, 2018-08-18:
Hi! So does this mean that for example with the ArduinoJson assistant if it shows me this:

```
const size_t bufferSize = JSON_OBJECT_SIZE(4) + 40;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "{\"m\":20000,\"s\":1,\"us\":0,\"t\":1534579769}";
```

If I use StaticJsonBuffer I can just do this instead?

```
const size_t bufferSize = JSON_OBJECT_SIZE(4);
StaticJsonBuffer<bufferSize> jsonBuffer;

const char* json = "{\"m\":20000,\"s\":1,\"us\":0,\"t\":1534579769}";
```

agathauy, 2018-08-18:
I'm now parsing the same thing:

```
	const size_t bufferSize2 = JSON_OBJECT_SIZE(4);
	StaticJsonBuffer<bufferSize2> jsonBuffer2;
	JsonObject& root2 = jsonBuffer2.parseObject(encoded);
```

But the parsing is still failing. It did free up the free memory between the stack and the heap to 136 though.

bblanchon, 2018-08-19:
Hi @agathauy,

Yes, your understanding is correct: the duplication only happens when the input is read-only.

The example you wrote should work; see this demo on wandbox:
https://wandbox.org/permlink/63sYyww4XBpDvEBJ

I guess that `freeMemory()` may not be very reliable.
Please try with a small program similar to the demo above to check that everything is fine.
Then backtrace to your original program until you trigger the bug again.

Regards,
Benoit

agathauy, 2018-08-19:
I think I may have found the problem. The text I'm parsing literally prints like this:
```
{\"m\":120000,\"s\":1,\"us\":0,\"t\":1534356559}
```
Instead of like this if one used ` char buffer[200]= "{\"m\":120000,\"s\":1,\"us\":0,\"t\":1534584738}"; `
```
{"m":120000,"s":1,"us":0,"t":1534584738}
```

And so I realized that from the demo you sent since apparently the data I was receiving also had the backslashes! I completely forgot that they weren't treated as escape characters anymore compared to when explicitly put in like this: `char buffer[200]= "{\"m\":120000,\"s\":1,\"us\":0,\"t\":1534584738}"; `.

Thank you so much for the help! :D And kudos on this awesome library too :D I got the parsing working already. I think the freeMemory() function I'm using is right since it's just subtracting the heap and stack. But anyways, thank you so much again! :D

closed 2018-08-19 by agathauy

bblanchon, 2018-08-21:
Hi @agathauy,

The demo on wandbox is often the trigger to see what's wrong in the code, that's why I always take the time to create one. I'm glad it worked for you too ;-)

Thanks for the update.
Good luck with your projects.

Regards,
Benoit