bblanchon/ArduinoJson #800: Prevent Sting when send reponse to server when using ESP8266WebServer

hasenradball, 2018-08-16:
Hello Benoit,

we discussed how to prevent a String when I want to send a response to the server.
```
void sendMessageJson(bool *pstate_relais, bool state_tor) {
  // sende die json message für den respone to client
  String json_str = "";
  // Memory pool for JSON object tree.
  StaticJsonBuffer<80> jsonBuffer;

  // Create the root of the object tree.
  JsonObject& root = jsonBuffer.createObject();

  root["Relais"] = *pstate_relais;
  root["Tor_offen"] = state_tor;
  //root.prettyPrintTo(Serial);
  //Serial.println();

  // es wird immer Relais Zustand mitgeteilt
  // wichtig! damit Daten nicht aus dem Browser cache kommen
  server.sendHeader("Cache-Control", "no-cache");
  // Antwort an Internet Browser senden --> bool (true/false
  root.printTo(json_str);
  server.send(200, "application/json", json_str);
}
```
The following proposal does not work directly:
```
void sendMessageJsonVar1(bool *pstate_relais, bool state_tor) {
  StaticJsonBuffer<80> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  root["Relais"] = *pstate_relais;
  root["Tor_offen"] = state_tor;

  WiFiClient& client = server.client();

  client.println("HTTP/1.0 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  root.printTo(client);
}
```
The following error occurs:

```
D:\Arduino\ESP8266\ESP8266_WiFi_Relais_Button_Json_StateMachine3_TEST\ESP8266_WiFi_Relais_Button_Json_StateMachine3_TEST.ino: In function 'void sendMessageJsonVar1(bool*, bool)':

ESP8266_WiFi_Relais_Button_Json_StateMachine3_TEST:231: error: invalid initialization of non-const reference of type 'WiFiClient&' from an rvalue of type 'WiFiClient'

   WiFiClient& client = server.client();
```

hasenradball, 2018-08-16:
Do you know the root cause?

hasenradball, 2018-08-17:
After a short research I will try the following:

WiFiClient client = server.available()

bblanchon, 2018-08-18:
Please try:

```c++
WiFiClient client = server.client();
```

hasenradball, 2018-08-18:
Hi,
I tried the following code, build was sucessfully but I receive an error in the html inspector.

The code:
```void sendMessageJson(bool *pstate_relais, bool state_tor) {
  StaticJsonBuffer<80> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  root["Relais"] = *pstate_relais;
  root["Tor_offen"] = state_tor;
  root.prettyPrintTo(Serial);
  WiFiClient client = server.client();

  client.println("HTTP/1.0 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  root.printTo(client);
}```

results in this on the Serial:
```{
  "Relais": false,
  "Tor_offen": true
}```

but on the Inspector of the HTML it prints:

``` [Mitteilungsdetails anzeigen/ausblenden] SyntaxError: JSON.parse: unexpected character at line 2 column 1 of the JSON data[Weitere Informationen]
```

hasenradball, 2018-08-18:
Arduino Build successful, but not in the Browser.

bblanchon, 2018-08-19:
[The response](https://en.wikipedia.org/wiki/Hypertext_Transfer_Protocol#Response_message) is missing an empty line between the headers and the body:

```c++
client.println("Connection: close");
client.println(); // <- HERE
root.printTo(client);
```

Please see [JsonServer.ino](https://arduinojson.org/v5/example/http-server/?utm_source=github&utm_medium=issues) for a complete example.

Regards,
Benoit

hasenradball, 2018-08-19:
This solution works:

```
void sendMessageJson(bool *pstate_relais, bool state_tor) {
  char json_str[256];

  StaticJsonBuffer<80> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  root["Relais"] = *pstate_relais;
  root["Tor_offen"] = state_tor;
  root.printTo(json_str);
  // wichtig! damit Daten nicht aus dem Browser cache kommen
  server.sendHeader("Cache-Control", "no-cache");
  server.send(200, "application/json", json_str);
}
```

hasenradball, 2018-08-19:
Solution #3 does also work:
```

void sendMessageJson(bool *pstate_relais, bool state_tor) {
  String json_str;
  json_str.reserve(256);

  StaticJsonBuffer<80> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  root["Relais"] = *pstate_relais;
  root["Tor_offen"] = state_tor;
  root.printTo(json_str);
  // wichtig! damit Daten nicht aus dem Browser cache kommen
  server.sendHeader("Cache-Control", "no-cache");
  server.send(200, "application/json", json_str);
}
```

hasenradball, 2018-08-19:
The Solution #1 does not work actually.
Solution #2 and #3 does work fine but which is the best one?

bblanchon, 2018-08-21:
Hi @hasenradball,

As I told you, I don't like the ESP8266WebServer because it forces you to use `String` with causes useless allocations and duplication.
I suggested to use the `WifiClient` to workaround this limitation but, apparently, it doesn't work.

With this information, I think that solution number 3 (the one using a `String` to store the JSON document and calling `reserve()` to reduce the number of allocation) represents the best compromise.
After all, it's just one short-lived heap allocation, which is perfectly acceptable on an ESP8266.

Regards,
Benoit

hasenradball, 2018-08-21:
Hi @bblanchon,

thanks for your answer.
If I have time in the next days I will try to use completely the ```WifiClient``` to check the first proposal. But for now I think I hav a good compromise.
Thank you very much!

bblanchon, 2018-08-22:
You're welcome @hasenradball.
Thank you for using ArduinoJson and good luck with your project.

closed 2018-08-22 by bblanchon

bblanchon, 2018-08-28:
Hi again @hasenradball,

As I told you, there is a caveat with using `WifiClient` directly with ArduinoJson.
It's captured in a [question in the FAQ](?utm_source=github&utm_medium=issues).

The code you have right now is a good compromise, so I recommend you stick with it.
Squishing the last bytes of memory is not required in your case because the ESP8266 has enough RAM.

Regards,
Benoit

hasenradball, 2018-08-30:
Hi Benoit,

I have one question concercning Strings. Lets assume I want write some data in the json object like this:
```
  root["Temperature"] = temperature;
  root["Humidity"] = humidity;
```
Ist it better to do it with the float values of temperature and humidity or convert these values as char array and do it like this:
```
  root["Temperature"] = temperatureStr;
  root["Humidity"] = humidityStr;
```

Regards
Frank

bblanchon, 2018-08-30:
If you have the floats variables, it's better to pass them to ArduinoJson.
It will use less RAM because the `JsonVariant` will store a `float` (4-bytes) instead of a copy of the string.
Also, the code will be simpler, the executable smaller, and probably faster.