bblanchon/ArduinoJson #801: StaticJsonBuffer was not declared in this scope (sorry for my bad English)

pku55, 2018-08-19:
Dear nerds

I'm faced with the "StaticJsonBuffer was not declared in this scope" problem.
I use an Arduino nanoESP and the ArduinoJson Library 5.13.2. I tested also other versions.
I do manage the libraries with the IDE. 
The Board without ISCP Connectors use the old bootloader (older than January 2018).
The IDE has the Version 1.8.5.
The included libraries are:
#include <SoftwareSerial.h>
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <ArduinoJson.h> 

Even the examples provided with the arduinoJson-master ZIP File can not be compiled an yield to 
"StaticJsonBuffer was not declared in this scope".

I would be very happy, if some knows a solution.
Many thanks Peter

pku55, 2018-08-21:
Many thanks Benoit for your reply.

I've attached two Files:
- Error message and Library Version (IDE Library Manager)
- The included Libraries, shows also the IDE Version
[image]
[image]
The program is written for ArduinoJson 5 and the included Library is 5.13.2. Not a Beta 6 Version.
Do you have any Idee how the problem can be solved?

Many thanks again Peter

pku55, 2018-08-21:
> Hello fellow human,
>
> This usually happens when a program written for ArduinoJson 5 is compiled against the version 6 of the library.
>
> Any chance that might be your case too?
>
> Regards,
> Benoit

Dear Benoit

 

Many thanks supporting me.

 

As you can see in the attached screen shots, the program is written for ArduinoJson 5 and the included library is ArduinoJson version 5.13.2, not a beta 6 version.

The IDE is 1.8.5 and the #include ArduinoJson.h command is done.

 

I would be very happy, if you have any ideas how to solve this case.

Thank you very much.

 

(I hope you understand my English)

 

Mit freundlichen Grüssen

Peter Kunz

Fritz-Käser-Strasse 18

4562 Biberist

+41 79 218 33 67

 <mailto:pku55@bluewin.ch> pku55@bluewin.ch

 <mailto:kunz-family@hotmail.com> kunz-family@hotmail.com

 

Sie sind eingeladen sich auf  <https://swissics.ch/> swissics.ch, unten auf den Seiten, für den Newsletter anzumelden und 
als Arbeitsuchender oder Arbeitgeber zu registrieren. Danke.

 

Success comes to those who fight for it

bblanchon, 2018-08-22:
Hi @pku55,

This is a very strange case...

Is it possible that there is another `ArduinoJson.h` installed on your computer?
It may be a manual installation from a zip file or a git clone.
I could also be one of the other installed libraries that contains an `ArduinoJson.h`.

BTW, don't apologize for your English; from what I can tell, it's much better than the average.

Regards,
Benoit

pku55, 2018-08-22:
Hi Benoit

 

Many thanks for your answer. 

 

I will check this. I do delete all arduinoJson Files and reinstall the library.

 

Mit freundlichen Grüssen

Peter Kunz

Fritz-Käser-Strasse 18

4562 Biberist

+41 79 218 33 67

 <mailto:pku55@bluewin.ch> pku55@bluewin.ch

 <mailto:kunz-family@hotmail.com> kunz-family@hotmail.com

 

Sie sind eingeladen sich auf  <https://swissics.ch/> swissics.ch, unten auf den Seiten, für den Newsletter anzumelden und 
als Arbeitsuchender oder Arbeitgeber zu registrieren. Danke.

 

Success comes to those who fight for it

pku55, 2018-08-22:
Hi Benoit

 

I think the problem is solved.

I do store all Arduino data on the disc (e:)

I consciously save all Arduino data on the disc (e:) 

I could find other Arduino libraries on disc (c:). I find out that the IDE will store them on (c:) 

So I have to chance the IDE presettings to get order in the “stable”.

Sorry, but I had to learn this first.

 

Many thanks for yout great support.

 

Mit freundlichen Grüssen

Peter Kunz

Fritz-Käser-Strasse 18

4562 Biberist

+41 79 218 33 67

 <mailto:pku55@bluewin.ch> pku55@bluewin.ch

 <mailto:kunz-family@hotmail.com> kunz-family@hotmail.com

 

Sie sind eingeladen sich auf  <https://swissics.ch/> swissics.ch, unten auf den Seiten, für den Newsletter anzumelden und 
als Arbeitsuchender oder Arbeitgeber zu registrieren. Danke.

 

Success comes to those who fight for it