bblanchon/ArduinoJson #802: memory leak 168 json.printTo()

anopid, 2018-08-21:
after each iteration of this code ESP8266 free heap get down with 168 bytes, there is no issue
if I use char array to store data.
code:
void getStatus() {
StaticJsonBuffer<bufferSize> jsonBuffer;
JsonObject &json = jsonBuffer.createObject();
json["gpio1"] = io[0].pin_state;
json["n1"] = io[0].io_name;
json["on1"] = io[0].io_name_on;
json["off1"] = io[0].io_name_off;
json["d1"] = io[0].ms_delay;
json["m1"] = io[0].pin_type;
...............
...............
json.printTo(Serial);
}

bblanchon, 2018-08-22:
Hi @anopid,

As far as I can tell, this function contains no memory leak; you can safely call `getStatus()` in a loop.
Any chance you might be in the same case as #795?

Regards,
Benoit

anopid, 2018-08-22:
Hi @bblanchon, thanks for info. After some tests, I can confirm that this leak is related with ESP8266 2.4.1 version of Print class, after update and test same code with 2.4.2 core there is no more memory leaks.

bblanchon, 2018-08-23:
Awesome!

closed 2018-08-23 by bblanchon