bblanchon/ArduinoJson #803: problem for updating json file every second

smartman2012, 2018-08-22:
Hi 
first of all, thank you for your great yet simple library.
I'm using JSON parser to pars status of some LEDs from my MySQL database (php+mysql+rest), everything is fine but there is a little problem ! when i update rows in the database using PHP, Arduino read JSON array but does not update the output of json immediately, it updates after 4-5 parsing.
can i clear json buffer before every time read json file from url ?
thank you

smartman2012, 2018-08-23:
Hello again
I checked codes and outputs again and find json is not the problem but HTTP request is the problem!
when I change the source and check in in google chrome update is made to data but ESP output does not change after 1-2 minutes or more (sometimes). how can I solve such a problem?
is it server side problem or client side?

bblanchon, 2018-08-28:
Hello @smartman2012,

I don't really understand the question.
Can you try to explain differently?
And maybe share a **small** code sample?

Regards,
Benoit

closed 2018-09-10 by bblanchon