bblanchon/ArduinoJson #804: Change key string

cmsavage, 2018-08-27:
Is it possible to change the key in a key-value pair?

I am working on a project where data will be logged to multiple locations locally and remotely.  Sometimes, that means passing data over an XBee network which has a limited packet payload size (<100 bytes).  Using descriptive keys is great for readability of log files, but cuts into the space available for data on a single XBee packet (avoiding some sort of multi-packet payload processing would be ideal).  To give myself a little more room to work with, I would like to replace some common keys with shorter 1-character labels when sending a JSON string over the XBee network and then reversing the process on the other end.  For example:
```json
{"Label":"foo", "Timestamp":1535399578, "Data":[1.234e5, 2.345e6, 3.456e7]}
```
would be changed to this:
```json
{"L":"foo", "T":1535399578, "D":[1.234e5, 2.345e6, 3.456e7]}
```
This could be achieved by removing the existing key-value pairs and adding them again under the new key name, but that seems a bit heavy-handed if it can be avoided.  Is it possible to simply change the key of a key-value pair without rewriting the value?

bblanchon, 2018-08-28:
Hi @cmsavage,

It's possible with version 5 when [iterating on the `JsonObject`](https://arduinojson.org/v5/api/jsonobject/begin_end/?utm_source=github&utm_medium=issues):

```c++
for( auto& kvp : obj) {
  kvp.key = "new key";
}
```

It's not possible in v6, and I don't think it will come back because it prevents me from organizing the object in a tree instead of a list.

Anyway, if you're concerned about size, maybe JSON is not the best representation.
I would suggest using a custom binary format or [MessagePack](https://arduinojson.org/v6/api/msgpack/serializemsgpack/?utm_source=github&utm_medium=issues), which is available on v6.

Regards,
Benoit

cmsavage, 2018-08-28:
Thanks, Benoit.

MessagePack does look like a useful possibility and, as far as I can tell, your library allows me to pack a JSON object (for logging) and MessagePack object (for network transfer) as one, and seamlessly convert the MessagePack back to JSON on the other end of the network.  I am definitely happy to avoid implementing my own custom format -- thanks for the suggestion!

closed 2018-09-10 by bblanchon