bblanchon/ArduinoJson #807: ESP8266 array of jsons

KristianDukov, 2018-09-04:
Hello All,

I am having the following code where I create a JSON every couple of seconds with some measurements:

`
char buffer[256];
root["reps"] = n;
root["power"] = powerContract;
root["weight"] = weight;
root["set"] = s;
root["contract"] = contract;
root["acc"] = "asd";
root["sensor"] = machine;
root.printTo(buffer);
String Json = String(buffer);
webSocket.broadcastTXT(Json);
`

Sending via WS works fine and the message is readable.

What I am trying to do is to create a list of all measurements I guess as an array so I can upload it via API.  I tried with

` JsonArray& array = jsonBuffer2.createArray();
JsonObject& buffer = array.createNestedObject();
`
but I guess I am missing something. Can somebody help?

THank you in advance

bblanchon, 2018-09-05:
Hi @KristianDukov,

It seems that you are in the right direction.
I wouldn't call the object `buffer`, but that should be a problem.
What's wrong with [`JsonArray::createNestedObject()`](https://arduinojson.org/v5/api/jsonarray/createnestedarray/?utm_source=github&utm_medium=issues)?

Regards,
Benoit

KristianDukov, 2018-09-05:
Hello,

I reviewed the output of JsonArray::createNestedObject() and I don't think it will work for me. I was thinking more to get an output such as:
'
{'sensor': 1, 'rep': 1}, {'sensor': 1, 'rep': 2}, '
So I ca post it to an API.

I will review what was the error I was getting tonight

Thank you!
Kris

KristianDukov, 2018-09-05:
Hello again,

I tried with the following code:
`
JsonArray& array = jsonBuffer.createArray();
array.add(buffer);
array.printTo(Serial);
`
And here is the output:

`
["{\"reps\":1,\"power\":3,\"weight\":80,\"set\":1,\"contract\":17725,\"account\":\"asd\",\"machine\":4}","{\"reps\":2,\"power\":46,\"weight\":80,\"set\":1,\"contract\":1267,\"account\":\"asd\",\"machine\":4}"]
`

Now, I am not sure if this is recognized as Json and if I can post it. My concern is with the  "\" character.

KristianDukov, 2018-09-05:
I think it worked out with

`
const char* jsonDuplicate = jsonBuffer.strdup(Json);
array.add(jsonBuffer.parseObject(jsonDuplicate));
array.printTo(Serial);
`

But how I can be sure that it will be recognized as separate JSONs? Does anybody know a website that I can post and test it?

bblanchon, 2018-09-06:
Hi @KristianDukov,

Usually, a JSON API that requires an array expects something like:

```json
[{"sensor": 1,"rep":1},{"sensor":1,"rep":2}]
```

...because that is how the JSON specification defines an array.
This is the kind of output you'll get if you call [JsonArray::printTo()](https://arduinojson.org/v5/api/jsonarray/printto/?utm_source=github&utm_medium=issues).

On the other hand, if you don't want the brackets at the beginning and the end, why are you using a [`JsonArray`](https://arduinojson.org/v5/api/jsonarray/?utm_source=github&utm_medium=issues)?
Instead, you could simply serialize each [`JsonObject`](https://arduinojson.org/v5/api/jsonobject/?utm_source=github&utm_medium=issues) and insert a comma between them.

```c++
obj1.printTo(Serial);
Serial.print(",");
obj2.printTo(Serial);
```

Regards,
Benoit

closed 2018-09-21 by bblanchon