bblanchon/ArduinoJson #808: get<String> returns "-0" for 0

RAlexeev, 2018-09-10:
`get<String>` returns different outputs of `0` for `v6.3.0-beta` and `v6.2.3-beta`. In case of `v6.3.0-beta` it returns `"-0"` (minus occurs before `0`), but in case of `v6.2.3-beta` it returns `"0"`.

Example:

```
char json[] = "{\"x\":0}";
StaticJsonDocument<256> doc;
deserializeJson(doc, json);
JsonObject object = doc.as<JsonObject>();
obj["x_str"] = object.get<String>("x");
obj["x_uint"] = object.get<unsigned int>("x");
```

Output in case of `v6.3.0-beta`:
```
x_str = "-0"
x_uint = 0
```
Output in case of `v6.2.3-beta`:
```
x_str = "0"
x_uint = 0
```

I use Platformio IDE 1.1.1 with the following settings:
```
platform = https://github.com/platformio/platform-espressif8266.git#v1.8.0
board = esp_wroom_02
framework = arduino
```

bblanchon, 2018-09-11:
Quite a funny bug :bug:
Nice catch!

bblanchon, 2018-09-11:
Fixed in [v6.4.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.4.0-beta)

closed 2018-09-11 by bblanchon