bblanchon/ArduinoJson #809: ARDUINOJSON_USE_DOUBLE corrupts output

RAlexeev, 2018-09-13:
I have some problems with `#define ARDUINOJSON_USE_DOUBLE 1` macros. See examples below.

### pi = 3

In this example I expect to get `{"pi":3}` instead of two different outputs like `{null:3}` and `{null:0}`
```
#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>

...

StaticJsonDocument<200> doc;
JsonObject root = doc.to<JsonObject>();
root["pi"] = 3;

serializeJson(doc, Serial);    // {null:3}
Serial.println("");

String jsonStr;
serializeJson(doc, jsonStr);
Serial.println(jsonStr);    // {null:0}
```

It seems strange that `serializeJson(doc, Serial);` and `serializeJson(doc, jsonStr); Serial.println(jsonStr);` produce different results.

If I comment out macros `#define ARDUINOJSON_USE_DOUBLE 1`, then I'll get correct result `{"pi":3}` in both type of outputs: `serializeJson(doc, Serial);` and `serializeJson(doc, jsonStr); Serial.println(jsonStr);`.

### pi = 3.14159265359

In this example I expect to get `{"pi":3.141592654}` instead of obtained `{null:3.141592654}` and exception.
```
#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>

...

StaticJsonDocument<200> doc;
JsonObject root = doc.to<JsonObject>();
root["pi"] = 3.14159265359;

serializeJson(doc, Serial);    // {null:3.141592654}
Serial.println("");

String jsonStr;
serializeJson(doc, jsonStr);    // Exception (see below)
Serial.println(jsonStr);
```

```
Exception (9):
epc1=0x40216e3d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4009220f depc=0x00000000

ctx: cont
sp: 3ffffb20 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffcc0:  3ffe8d95 3fff30ac 00000000 00000000
3ffffcd0:  40000000 3ffed650 3ffffe98 40215167
3ffffce0:  3ffe8da4 00000001 3ffe8d91 00000000
3ffffcf0:  3ffe8da4 3ffffe70 3ffffe8c 40215234
3ffffd00:  3ffe002c 3ffffda4 3ffffe8c 4020f09e
3ffffd10:  fffffffe 00000000 3ffffe8c 4020ed9a
3ffffd20:  3ffe8da4 3ffffe70 3ffffe8c 4020f8b1
3ffffd30:  00000000 3ffffd94 3ffffda4 3ffffd94
3ffffd40:  3ffffda4 3ffffd94 400921fb 402143b1
3ffffd50:  3ffffd94 3ffffe74 3ffffe98 402143b1
3ffffd60:  3ffe92e2 3ffffe98 3ffffe8c 4020f0f0
3ffffd70:  3ffffd94 3ffffe74 3ffef828 402145d8
3ffffd80:  3ffe8da4 3ffffe98 3ffef828 402134cb
3ffffd90:  0000000a 40217608 3ffffda4 000000c8
3ffffda0:  00000020 0000000a 00000000 54442eea
3ffffdb0:  400921fb 00000000 00000000 3ffe9361
3ffffdc0:  00000000 40214dd7 0000010a 3ffef968
3ffffdd0:  00000010 00000010 00000000 40100579
3ffffde0:  00000010 40214dd7 00000000 3ffef67c
3ffffdf0:  00000010 3fff1a8c 3fff1a8c 40214dd7
3ffffe00:  3ffe8da4 00000000 3fff1a8c 40214e23
3ffffe10:  00000010 00000010 3fff1a8c 40214e55
3ffffe20:  3ffe8da4 3fff1a80 3fff1a8c 40214ea2
3ffffe30:  3ffe8da4 40215840 0000001c 401004e8
3ffffe40:  3fff2144 4020c139 3ffe8da4 4020bf38
3ffffe50:  00000010 3fff1a80 3ffef5d0 3ffef968
3ffffe60:  00000010 00000010 00000000 40100579
3ffffe70:  00000009 3ffffda4 3ffffda4 3ffef968
3ffffe80:  3ffffd94 3ffffe70 00000000 3ffffeac
3ffffe90:  00000003 40214dd7 3fff1e04 0000000f
3ffffea0:  00000008 3ffffd94 3ffffe74 3ffffe98
3ffffeb0:  3ffffe70 00000394 00000394 4010020c
3ffffec0:  00000001 00000001 3fffff00 4020bb0c
3ffffed0:  00000001 00000001 3fff19ec 40216f7e
3ffffee0:  00000000 00000000 3fff19ec 4020bb02
3ffffef0:  3fff19ec 3ffef614 3fffff20 4020bb3e
3fffff00:  00000000 00000000 00000000 40214f80
3fffff10:  3fff19ec 3ffef614 3ffef5d0 4020bbc5
3fffff20:  3fff2174 0000000f 00000007 00000000
3fffff30:  00000000 4bc6a7f0 0000979b 3ffef8e8
3fffff40:  3ffef614 00000001 3ffe85dc 3ffef5f8
3fffff50:  00000001 00000000 4020aeb4 0000000d
3fffff60:  00000000 3fff1dcc 3ffef5d0 3ffef8e8
3fffff70:  00000001 3ffef5f8 3ffef5d0 4020be00
3fffff80:  402174e8 00000000 00001388 40213868
3fffff90:  00000000 3fff1dcc 40216f88 40216f74
3fffffa0:  3fffdad0 00000000 3ffef8e0 4021388a
3fffffb0:  3fffdad0 00000000 3ffef8e0 402159c0
3fffffc0:  feefeffe feefeffe 3ffe85dc 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```

If I comment out macros `#define ARDUINOJSON_USE_DOUBLE 1`, then I'll get correct result `{"pi":3.141593}` in both type of outputs: `serializeJson(doc, Serial);` and `serializeJson(doc, jsonStr); Serial.println(jsonStr);`.

I use ArduinoJson 6.4.0-beta and compile program by Platformio IDE 1.1.1 with the following settings:
```
platform = https://github.com/platformio/platform-espressif8266.git#v1.8.0
board = esp12e
framework = arduino
```

bblanchon, 2018-09-13:
Hi @RAlexeev,

Thank you for this clear explanation of the problem.

I just made a test, and it seems to work as expected.
See by yourself: https://wandbox.org/permlink/Q42jITWzXNKI50v5

However, I've seen similar stuff happen when there are multiple compilation units with different configurations.
For example, if `a.cpp` and `b.cpp` define different values for `ARDUINOJSON_USE_DOUBLE`, the compiled executable could behave like that.

So, before investigating further, I want to make sure that you checked that all compilation units agrees on the value of `ARDUINOJSON_USE_DOUBLE`.

Regards,
Benoit

RAlexeev, 2018-09-14:
@bblanchon, thank you very much for your explanation! You're absolutely right. I've added macros `#define ARDUINOJSON_USE_DOUBLE 1` to another source file, which include library `ArduinoJson`, and now program works properly. Anyway, I'm sorry for disturbing you.

bblanchon, 2018-09-14:
No worries.
I'll try to allow mixed configurations in future versions.
Meanwhile, I'll keep this issue opened as a reminder.